basler/pypylon #240: Mac OS pip install failure due to filename typo

dbricare, 2020-05-19:
When cloning the pypylon repo locally and using `pip install .` I receive an error.

The issue is due to a mismatch between the file created in the pylon 6.1.2 Mac OS package and pypylon file [setup.py line 687](https://github.com/basler/pypylon/blob/master/setup.py#L687)

A quick work-around involves copying the file with a new name
```
cp /Library/Frameworks/pylon.framework/Versions/Current/Resources/Tools/pylon-config
 /Library/Frameworks/pylon.framework/Versions/Current/Resources/Tools/pylon-config.sh
```
Rather than submit a pull request I figured this could easily be fixed by the dev team during their next commit. Thanks.

```
    ERROR: Command errored out with exit status 1:
     command: /usr/local/opt/python/bin/python3.7 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/var/folders/90/m73vsm9j65v09n2tyb5knn980000gn/T/pip-req-build-bc2pt03s/setup.py'"'"'; __file__='"'"'/private/var/folders/90/m73vsm9j65v09n2tyb5knn980000gn/T/pip-req-build-bc2pt03s/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /private/var/folders/90/m73vsm9j65v09n2tyb5knn980000gn/T/pip-pip-egg-info-la04ol2m
         cwd: /private/var/folders/90/m73vsm9j65v09n2tyb5knn980000gn/T/pip-req-build-bc2pt03s/
    Complete output (20 lines):
    ERROR:root:Couldn't find pylon. Please install pylon in /Library/Frameworks/ or tell us the installation location using the PYLON_ROOT environment variable
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/private/var/folders/90/m73vsm9j65v09n2tyb5knn980000gn/T/pip-req-build-bc2pt03s/setup.py", line 814, in <module>
        default=bs.get_version(),
      File "/private/var/folders/90/m73vsm9j65v09n2tyb5knn980000gn/T/pip-req-build-bc2pt03s/setup.py", line 285, in get_version
        pylon_version = self.get_pylon_version()
      File "/private/var/folders/90/m73vsm9j65v09n2tyb5knn980000gn/T/pip-req-build-bc2pt03s/setup.py", line 766, in get_pylon_version
        return self.call_pylon_config("--version")
      File "/private/var/folders/90/m73vsm9j65v09n2tyb5knn980000gn/T/pip-req-build-bc2pt03s/setup.py", line 748, in call_pylon_config
        res = subprocess.check_output(params, universal_newlines=True)
      File "/usr/local/Cellar/python/3.7.7/Frameworks/Python.framework/Versions/3.7/lib/python3.7/subprocess.py", line 411, in check_output
        **kwargs).stdout
      File "/usr/local/Cellar/python/3.7.7/Frameworks/Python.framework/Versions/3.7/lib/python3.7/subprocess.py", line 488, in run
        with Popen(*popenargs, **kwargs) as process:
      File "/usr/local/Cellar/python/3.7.7/Frameworks/Python.framework/Versions/3.7/lib/python3.7/subprocess.py", line 800, in __init__
        restore_signals, start_new_session)
      File "/usr/local/Cellar/python/3.7.7/Frameworks/Python.framework/Versions/3.7/lib/python3.7/subprocess.py", line 1551, in _execute_child
        raise child_exception_type(errno_num, err_msg, err_filename)
    FileNotFoundError: [Errno 2] No such file or directory: '/Library/Frameworks/pylon.framework/Versions/Current/Resources/Tools/pylon-config.sh': '/Library/Frameworks/pylon.framework/Versions/Current/Resources/Tools/pylon-config.sh'
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
```

thiesmoeller, 2020-05-25:
Thanks for reporting.
The release of a new pylon version changed the name of the config script

SanliFaez, 2020-08-27:
I have encountered the same issue with a recent try to install on Mac and the work-around did not fix it for me
`    ERROR: Command errored out with exit status 1:
     command: /Library/Frameworks/Python.framework/Versions/3.6/bin/python3.6 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/var/folders/hx/sr3d3rq11r36vxdb56cmsvvh0000gn/T/pip-req-build-hi7s_ei7/setup.py'"'"'; __file__='"'"'/private/var/folders/hx/sr3d3rq11r36vxdb56cmsvvh0000gn/T/pip-req-build-hi7s_ei7/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /private/var/folders/hx/sr3d3rq11r36vxdb56cmsvvh0000gn/T/pip-pip-egg-info-ugtv94cm
         cwd: /private/var/folders/hx/sr3d3rq11r36vxdb56cmsvvh0000gn/T/pip-req-build-hi7s_ei7/
    Complete output (11 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/private/var/folders/hx/sr3d3rq11r36vxdb56cmsvvh0000gn/T/pip-req-build-hi7s_ei7/setup.py", line 818, in <module>
        bs = BuildSupport.make()
      File "/private/var/folders/hx/sr3d3rq11r36vxdb56cmsvvh0000gn/T/pip-req-build-hi7s_ei7/setup.py", line 332, in make
        return BuildSupportMacOS()
      File "/private/var/folders/hx/sr3d3rq11r36vxdb56cmsvvh0000gn/T/pip-req-build-hi7s_ei7/setup.py", line 725, in __init__
        self.SwigExe = self.find_swig()
      File "/private/var/folders/hx/sr3d3rq11r36vxdb56cmsvvh0000gn/T/pip-req-build-hi7s_ei7/setup.py", line 128, in find_swig
        raise RuntimeError("swig executable not found on path!")
    RuntimeError: swig executable not found on path!
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
`

thiesmoeller, 2020-08-27:
You need swig as described in readme.

It is reported at the end of your message
`RuntimeError: swig executable not found on path!`

SanliFaez, 2020-08-28:
thank you! 
indeed, installing swig allowed me to move forward, only to encounter the same error as the beginning of thread. after adding the .sh extension to pylon-config, I could run pip install . which worked mostly fine (copying and building) but was terminated with:
```
     warning: include path for stdlibc++ headers not found; pass '-stdlib=libc++' on the command line to use the libc++ standard library instead [-Wstdlibcxx-not-found]
    /private/var/folders/hx/sr3d3rq11r36vxdb56cmsvvh0000gn/T/pip-req-build-zjs6zmcd/generated/genicam_wrap.cpp:2749:10: fatal error: 'string' file not found
    #include <string>
             ^~~~~~~~
    1 warning and 1 error generated.
    error: command '/usr/bin/clang' failed with exit status 1
```

and thank you for supporting pypylon, we use it on our MS-Win PCs mostly out of the box.

thiesmoeller, 2020-08-28:
`include path for stdlibc++ headers not found`

I have no issue building 
In homebrew environment 
Your clang seems to have setup issue

commit d4c921fe (2020-10-20): Fix name of python-config script on macOS. Resolves #240.
closed 2020-10-20 by stefanklug