basler/pypylon #241: Update Readme to clarify which .whl to download

CDahmsCellarEye, 2020-05-20:
Long time Pylon user but 1st time PyPylon user here.  Somebody please update the Readme to explain which .whl to download.  Currently the Readme provides a link to the releases page https://github.com/basler/pypylon/releases however there are many .whl files for each release and it's not clear which is applicable for certain hardware.

For example for the 1.5.4 release there are 33 .whls available.  I'm using a 64-bit "x86" (i.e. non-ARM) computer running Ubuntu 18.04.  So if I eliminate the `i686` (meaning 32 bit), `aarch64` (meaning ARM 64-bit), and `win` (meaning windows) I'm left with the following 9 options:

```
pypylon-1.5.4-cp35-cp35m-manylinux2014_x86_64.whl
pypylon-1.5.4-cp36-cp36m-manylinux2014_x86_64.whl
pypylon-1.5.4-cp37-cp37m-manylinux2014_x86_64.whl
pypylon-1.5.4-cp38-cp38-manylinux2014_x86_64.whl
pypylon-1.5.4-cp34-cp34m-linux_x86_64.whl
pypylon-1.5.4-cp35-cp35m-linux_x86_64.whl
pypylon-1.5.4-cp37-cp37m-linux_x86_64.whl
pypylon-1.5.4-cp38-cp38-linux_x86_64.whl
pypylon-1.5.4-cp36-cp36m-linux_x86_64.whl
```
What are the `cpXX` and `cpXXm` numbers?  What is the difference between `linux` and `manylinux`?

I'm under the impression that the increasing numbers are more recent builds, so that has me guessing between `pypylon-1.5.4-cp38-cp38-linux_x86_64.whl` and `pypylon-1.5.4-cp38-cp38-manylinux2014_x86_64.whl`.

Please update the Readme to clarify this.  Also I tried the pip3 package via:

`$ sudo -H pip3 install pypylon`

and received the following error:

```
ERROR: Could not find a version that satisfies the requirement pypylon (from versions: none)
ERROR: No matching distribution found for pypylon
```

Do I need a special pip3 command or is the Linux .whl not available via pip3?

sinasartip, 2020-06-04:
So i believe the cpxx is your python version. So cp37 is python 3.7.x i think this is just a common convention of writing python versions. 
In terms of which to install i downloaded the many linux for my ubuntu system running python 3.8.

although with all this said if you actually try and install the wrong one the program will give you an error so you know you have the wrong one.

thiesmoeller, 2025-12-15:
Closing as enhancement request. The cpXX numbers indicate Python version (cp38 = Python 3.8). manylinux wheels are preferred for better compatibility. For Ubuntu 18.04 x86_64 with Python 3.8, use: pypylon-X.X.X-cp38-cp38-manylinux2014_x86_64.whl. PyPI doesn't host pypylon wheels - download from GitHub releases. The README could be clearer about this, and we'll consider improving it. For now, pip install won't work - you must download the wheel file manually. This has been greatly enhanced by providing stable ABI builds since pypylon 4.x