basler/pypylon #243: EventSelector_Line1RisingEdge

Cody-G, 2020-06-03:
I see in the SDK documentation that there are trigger line edge events, for example `EventSelector_Line1RisingEdge `, and that these are supposed to apply to GigE cameras.  However I am unable to enable these events through the Python API or the Pylon viewer (they don't show up as options for the event selector).

Is it possible to use these events with a GigE camera?  My goal is to register a callback that runs each time a particular line gets triggered, and to save the camera timestamp of the trigger event.  Perhaps there's another way to do this, without using these events?  Thanks!

thiesmoeller, 2020-06-04:
If you look up the documentation on docs.baslerweb.com this feature is not available on your model.

Alternative would be:
* Polling ( not fast not precise. Can be precise if you sync time between camera and host with ieee1588 )
* If IO status in moment of frame  capture is required: use "Line Status All Chunk"

Cody-G, 2020-06-05:
Thanks for your quick response!  I saw the event node in the Pylon API and was hoping that my camera might (unofficially) support this event.  In my case I think neither polling nor checking the IO status is ideal.

Can I estimate a timestamp for the hardware trigger event by subtracting like this? 

`GrabResult.ChunkTimestamp - TriggerDelayAbs - exposure_start_delay`, where `exposure_start_delay` is as defined [here](https://docs.baslerweb.com/acquisition-timing-information.html#specifics) in the Basler docs?

thiesmoeller, 2025-12-15:
This issue asks about using trigger line edge events with GigE cameras. The question appears to have been addressed in the comments. We are closing this issue to keep our tracker current. If you still need assistance with event selectors, please open a new issue with your specific camera model and use case.