basler/pypylon #245: disable white balancing and all other auto adjustments for daA4200-30mci

L-o-r-e-n-z, 2020-06-03:
Hello all,

in February you updated the driver packages for the daA4200-30mci camera type. There you implemented the exposure and gain adjustment possibility and to also disable the auto adjustment. Is it possible that you also implement other auto adjustments or especially the disabling of this? How is the chance to implement those features soon?

Thank you for your support

FrankKarstens, 2020-06-08:
Hi L-o-r-e-n-z,
we regularly implement new features on customer requests but prioritized by the feature's usefulness for the majority of our customers. Maybe we can discuss your needs offline: mailto:frank.karstens@baslerweb.com

stefanklug, 2020-08-26:
Discussion moved to other channel. 
@L-o-r-e-n-z  we added manual white balance in the current software (CEP 1.4)

closed 2020-08-26 by stefanklug

HosseinSD, 2021-01-28:
I also need to disable auto white balance, is this feature added yet?