basler/pypylon #247: Using "FeaturePersistence" for Gige Stereo rig.

DylanDC-C, 2020-06-17:
As the title says, I want to load a configuration file into two Basler gige cameras. For a single camera the code works with no issue:

```
import os

os.environ["PYLON_CAMEMU"] = "3

from pypylon import pylon
from pypylon import genicam
import sys

exitcode = 0

countOfImagesToGrab = 5

try:
    camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
    camera.Open()
    pylon.FeaturePersistence.Load('Config.txt', camera.GetNodeMap(), True)
    print("Using device: ", camera.GetDeviceInfo().GetModelName())

    camera.StartGrabbingMax(countOfImagesToGrab)
   
    ......

sys.exit(exitcode)
```

But adding the same code to the stereo sample gives the following error:

```
import os

os.environ["PYLON_CAMEMU"] = "3"

from pypylon import pylon
from pypylon import genicam
import sys

exitCode = 0

countOfImagesToGrab = 2
maxCamerasToUse = 2

try:
    tlFactory = pylon.TlFactory.GetInstance()

    devices = tlFactory.EnumerateDevices()
    if len(devices) ==0:
        raise pylon.RuntimeException("No Camera present.")

    cameras = pylon.InstantCameraArray(min(len(devices),maxCamerasToUse))

    l = cameras.GetSize()
    for i, cam in enumerate(cameras):
        cam.Attach(tlFactory.CreateDevice(devices[i]))
        print("Using device ",cam.GetDeviceInfo().GetModelName())
        pylon.FeaturePersistence.Load('Config.txt', cam.GetNodeMap(), True)

    cameras.StartGrabbing()

    .....

sys.exit(exitCode)
```

> Traceback (most recent call last):
  File "StereoCapture.py", line 40, in <module>
    pylon.FeaturePersistence.Load('Config.txt', cam.GetNodeMap(), True)
  File "venv\lib\site-packages\pypylon\pylon.py", line 7042, in Load
    return _pylon.FeaturePersistence_Load(FileName, pNodeMap, validate)
_genicam.AccessException: Node is not writable. : AccessException thrown in node 'DeviceRegistersStreamingStart' while calling 'DeviceRegistersStreamingStart.Execute()' (file 'commandt.h', line 61)

From this I get that within the `for i, cam in enumerate(cameras)` loop, the nodemap is not accessible. So is there a way around this to make the nodemap writable or is there a better way in general to do this.

thiesmoeller, 2020-06-17:
Camera.Open() is missing

DylanDC-C, 2020-06-18:
Thank you, it works fine now. I feel like an idiot, the code could run without the `.Open()` module so I thought it was handled in the background.

closed 2020-06-18 by DylanDC-C