basler/pypylon #248: Is the MultiCast working with pypylon? (I cannot find sample for Grab_MultiCast.py)

tcolov, 2020-06-22:
MultiCast feature make the difference for me with Basler Cams. I'm using at the moment acA780-75gc model.

MultiCast sample is available in Pylon SDK (/opt/pylon/share/pylon/Samples/C++/Grab_MultiCast/) in C++ source code but I need it in python.

What is the right way to set with pypylon the following C++ example below: 
            camera.GetStreamGrabberParams().TransmissionType = TransmissionType_Multicast;

thiesmoeller, 2020-06-22:
camera.StreamGrabber.Transmissiontype = 'Multicast'

tcolov, 2020-06-23:
Could you please convert the Pylon MultiCast sample code from C++ '/opt/pylon/share/pylon/Samples/C++/Grab_MultiCast/Grab_MultiCast.cpp' to Pypylon sample code: Grab_MultiCast.py  
( may be some of the MultiCast Pylon APIs are missing in the current  Pypylon release? )

###############################################
I'm using Nvidia Jetson Nano with Pylon SDK version 6.1.0.19674-deb0
###############################################
root@jetson-nano:/opt/pylon/include# dpkg -l |grep pylon
ii  pylon                                         6.1.0.19674-deb0                                 arm64        Basler pylon Camera Software Suite
ii  pylon-supplementary-package-for-mpeg-4        1.0.1.117-deb0                                   arm64        Basler pylon Supplementary Package for MPEG-4
root@jetson-nano:/opt/pylon/include# cat /etc/nv_tegra_release 
# R32 (release), REVISION: 3.1, GCID: 18186506, BOARD: t210ref, EABI: aarch64, DATE: Tue Dec 10 06:58:34 UTC 2019
root@jetson-nano:/opt/pylon/include# uname -a
Linux jetson-nano 4.9.140-tegra #1 SMP PREEMPT Mon Dec 9 22:47:42 PST 2019 aarch64 aarch64 aarch64 GNU/Linux
#########################################

############################
Here is the error output I get with the usage of:
camera.StreamGrabber.Transmissiontype = 'Multicast'

Output: 
root@jetson-nano:/usr/src/pypylon/pypylon/samples# python3  Grab_MultiCast.py
#################################
Using device  Basler acA780-75gc#003053265F32#192.168.10.11:3956
An exception occurred.
Traceback (most recent call last):
  File "Grab_MultiCast.py", line 38, in <module>
    camera.StreamGrabber.Transmissiontype = 'Multicast'
  File "/usr/local/lib/python3.6/dist-packages/pypylon/genicam.py", line 1511, in __setattr__
    self.GetNode(attribute).SetValue(val)
  File "/usr/local/lib/python3.6/dist-packages/pypylon/genicam.py", line 1473, in GetNode
    return _genicam.INodeMap_GetNode(self, Name)
_genicam.LogicalErrorException: Node not existing (file 'genicam_wrap.cpp', line 16603)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "Grab_MultiCast.py", line 63, in <module>
    print(e.GetDescription())
AttributeError: 'LogicalErrorException' object has no attribute 'GetDescription'

##########################
Here is the Code in pypylon:
###################################3
from pypylon import pylon
from pypylon import genicam

import sys

# Number of images to be grabbed.
countOfImagesToGrab = 100

# The exit code of the sample application.
exitCode = 0

try:
    # Create an instant camera object with the camera device found first.
	ip_address = "192.168.10.11"
	info = pylon.DeviceInfo()
	info.SetPropertyValue('IpAddress',ip_address)
    

	camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice(info))

	print("Using device ", camera.GetDeviceInfo().GetFullName())
	camera.Open()

	camera.StreamGrabber.Transmissiontype = 'Multicast'

	camera.StartGrabbingMax(countOfImagesToGrab)

    # Camera.StopGrabbing() is called automatically by the RetrieveResult() method
    # when c_countOfImagesToGrab images have been retrieved.
	while camera.IsGrabbing():
        # Wait for an image and then retrieve it. A timeout of 5000 ms is used.
		grabResult = camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)

        # Image grabbed successfully?
		if grabResult.GrabSucceeded():
            # Access the image data.
			print("SizeX: ", grabResult.Width)
			print("SizeY: ", grabResult.Height)
			img = grabResult.Array
			print("Gray value of first pixel: ", img[0, 0])
		else:
			print("Error: ", grabResult.ErrorCode, grabResult.ErrorDescription)
		grabResult.Release()
	camera.Close()

except genicam.GenericException as e:
    # Error handling.
	print("An exception occurred.")
	print(e.GetDescription())
	exitCode = 1

sys.exit(exitCode)

################################################
Here is the output without trying to set MultiCast feature:
################################################
Using device  Basler acA780-75gc#003053265F32#192.168.10.11:3956
SizeX:  224
SizeY:  224
Gray value of first pixel:  4
SizeX:  224
SizeY:  224
Gray value of first pixel:  4

thiesmoeller, 2020-06-23:
@tcolov ... fixed typo in last comment.

.... All features names in C++ are 1:1 in python 
.... If you post code to issues please use the correct markdown "insert code"

tcolov, 2020-06-23:
Unfortunately the sample C++ codes are not converted 1:1 in pypylon python. The MultiCast example is missing.
Who can help with the converting the  /opt/pylon/share/pylon/Samples/C++/Grab_MultiCast.cpp to Grab_MultiCast.py ?

My attempt failed with error:
 _genicamLogicalErrorException: **Node not existing** (file '**genicam_wrap.cpp',** line 16603).
( above in this thread you can find the sample code I used )

tcolov, 2020-06-23:
It is working :) thanks for the support.
Transmissiontype - should have been with capital T in "type" :  TransmissionType

##########################################
Here is the sample code I successfully used to try it: 
##########################################
`from pypylon import pylon

tl_factory = pylon.TlFactory.GetInstance()

cam = None
for dev_info in tl_factory.EnumerateDevices():
    if dev_info.GetDeviceClass() == 'BaslerGigE':
        print("using %s @ %s" % (dev_info.GetModelName(), dev_info.GetIpAddress()))
        cam = pylon.InstantCamera(tl_factory.CreateDevice(dev_info))
        break
else:
    raise EnvironmentError("no GigE device found")

cam.Open()
print("AcquisitionMode", cam.AcquisitionMode.GetValue())
cam.StreamGrabber.TransmissionType = 'Multicast'
countOfImagesToGrab = 10000
cam.StartGrabbingMax(countOfImagesToGrab)
while cam.IsGrabbing():
        # Wait for an image and then retrieve it. A timeout of 5000 ms is used.
	grabResult = cam.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)

        # Image grabbed successfully?
	if grabResult.GrabSucceeded():
            # Access the image data.
		print("SizeX: ", grabResult.Width)
		print("SizeY: ", grabResult.Height)
		img = grabResult.Array
		print("Gray value of first pixel: ", img[0, 0])
	else:
		print("Error: ", grabResult.ErrorCode, grabResult.ErrorDescription)
	grabResult.Release()

cam.Close()
`

closed 2020-06-23 by tcolov