basler/pypylon #249: change bit depth

niallodowd, 2020-06-22:
Hey PyPylon! I was wondering how to change the bit -depth on my camera. Is there a command where I can change my bit depth from 8 to 12? Also, if there is a method that I should use to figure out what commands are available for my object class "cam" (after `cam = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())`) please let me know!

Thanks,

thiesmoeller, 2020-06-22:
Hi @niallodowd ,

the features to access from "cam" are dynamically loaded from the camera. This is the genicam technology.

`print(dir(cam))` would show all.

Please go to docs.baslerweb.com or just open the pylon viewer application. All features that you see are accessible via pypylon.

To switch depth. Change feature PixelFormat.

niallodowd, 2020-06-22:
Thanks so much for help! I'm still learning the basics, and it looks like when I try

`camera.PixelFormat=PixelType_BayerBG12` i get an error specifying that I need to assign a string to the pixel format....?

niallodowd, 2020-06-22:
I also tried:

` camera.Open()
camera.PixelFormat = "BayerBG12"
camera.close()`

to recieve an error: **Node is not writable : AccessException thrown in node 'PixelFormat' while calling 'PixelFormat.FromString()' (file 'valuet.h', line 85)**

niallodowd, 2020-06-22:
I was able to figure this issue out: I was grabbing frames while trying to change the pixelformat! Thanks for your initial comment.

stefanklug, 2020-08-26:
Great that the issue got solved :-)

closed 2020-08-26 by stefanklug