basler/pypylon #250: the constants of pylon.OutputOrientation_xxx seem wrong

zenyusy, 2020-06-24:
Hello all! When using the attrib `OutputOrientation` of `pypylon.pylon.ImageFormatConverter` to manipulate image orientation,
I found the built-in integer values provided seemed wrong. 

Please run the demo below:

```python
from pypylon import pylon

def try_set_orientation(dct, msg):

    print(f'\nSet output orientation by {msg} constants\n')

    for k, v in dct.items():
        cvt = pylon.ImageFormatConverter()
        print(f'Init orientation: {cvt.OutputOrientation.GetValue()}')
        try:
            print(f'Using {v} to set {k}')
            cvt.OutputOrientation = v
        except Exception as e:
            print(f'Set orientation err: {e}')
        finally:
            print(f'Final orientation: {cvt.OutputOrientation.GetValue()}\n')

pylon_const = {
        'Unchanged': pylon.OutputOrientation_Unchanged,  # = 2
        'TopDown':   pylon.OutputOrientation_TopDown,  # = 1
        'BottomUp':  pylon.OutputOrientation_BottomUp  # = 0, which triggers InvalidArgumentException
        }

foo_const = {
        'Unchanged': 1,
        'TopDown':   2,
        'BottomUp':  3
        }

if __name__ == '__main__':
    try_set_orientation(pylon_const, 'pypylon built-in')
    try_set_orientation(foo_const, 'some different')
```

On my computer with **pypylon 1.5.4** installed (*via pip*), the 3 values are all incorrect as the inline comments show

anders-ae-dahl, 2020-09-12:
I can verify this bug. Adding +1 to all enums works as a workaround.

thiesmoeller, 2020-09-12:
thanks for reporting. Currently searching for a good solution on swig level.

 workaround is to use string assignment.

`cvt.OutputOrientation = "BottomUp"`