basler/pypylon #251: Save images in different time steps

drexlrosh, 2020-07-05:
Hello all,

since I'm very new to python i would really appreciate your help regarding my issue. I'm using the code below to save five images from my basler camera to my drive.

Now I want to save each image every 10 seconds (saving each image as a file to my drive). Example: First image after 0 sec, second image after 10 sec, third image after 20 sec and so on. How could I develop such code?

Your help is much appreciated.

Best regards

```
from pypylon import pylon
import platform

num_img_to_save = 5
img = pylon.PylonImage()
tlf = pylon.TlFactory.GetInstance()

cam = pylon.InstantCamera(tlf.CreateFirstDevice())
cam.Open()
cam.StartGrabbing()
for i in range(num_img_to_save):
    with cam.RetrieveResult(2000) as result:

        # Calling AttachGrabResultBuffer creates another reference to the
        # grab result buffer. This prevents the buffer's reuse for grabbing.
        img.AttachGrabResultBuffer(result)

        if platform.system() == 'Windows':
            # The JPEG format that is used here supports adjusting the image
            # quality (100 -> best quality, 0 -> poor quality).
            ipo = pylon.ImagePersistenceOptions()
            quality = 90 - i * 10
            ipo.SetQuality(quality)

            filename = "saved_pypylon_img_%d.jpeg" % quality
            img.Save(pylon.ImageFileFormat_Jpeg, filename, ipo)
        else:
            filename = "saved_pypylon_img_%d.png" % i
            img.Save(pylon.ImageFileFormat_Png, filename)

        # In order to make it possible to reuse the grab result for grabbing
        # again, we have to release the image (effectively emptying the
        # image object).
        img.Release()

cam.StopGrabbing()
cam.Close()
```

thiesmoeller, 2020-07-07:
you can sleep between the retrieve_results ...
and you have to tell the grab engine, that you are only interested in the freshest images ..
```
from pypylon import pylon
import platform
import time

num_img_to_save = 5
img = pylon.PylonImage()
tlf = pylon.TlFactory.GetInstance()

cam = pylon.InstantCamera(tlf.CreateFirstDevice())
cam.Open()
cam.StartGrabbing( pylon.GrabStrategy_LatestImageOnly)
for i in range(num_img_to_save):
    with cam.RetrieveResult(2000) as result:

        # Calling AttachGrabResultBuffer creates another reference to the
        # grab result buffer. This prevents the buffer's reuse for grabbing.
        img.AttachGrabResultBuffer(result)

        if platform.system() == 'Windows':
            # The JPEG format that is used here supports adjusting the image
            # quality (100 -> best quality, 0 -> poor quality).
            ipo = pylon.ImagePersistenceOptions()
            quality = 90 - i * 10
            ipo.SetQuality(quality)

            filename = "saved_pypylon_img_%d.jpeg" % quality
            img.Save(pylon.ImageFileFormat_Jpeg, filename, ipo)
        else:
            filename = "saved_pypylon_img_%d.png" % i
            img.Save(pylon.ImageFileFormat_Png, filename)

        # In order to make it possible to reuse the grab result for grabbing
        # again, we have to release the image (effectively emptying the
        # image object).
        img.Release()
    time.sleep(10)
cam.StopGrabbing()
cam.Close()

```

stefanklug, 2020-08-25:
I'm closing this, as there was no activity in the last month. If needed please comment & reopen.

closed 2020-08-25 by stefanklug

thasnimolvs, 2021-01-26:
how to grab and save images in basler camera using threading