basler/pypylon #253: Error: 3792764924 UX Status: Libusb error: **UNKNOWN*

Neel1302, 2020-07-09:
Hi,
I have 4 basler aca2040-55uc USB3 cameras connected directly to the USB3 ports on an AMDx399 based chipset motherboard (https://www.gigabyte.com/ca/Motherboard/X399-DESIGNARE-EX-rev-10#kf). I am using Ubuntu 18.04 (kernel: 5.6.7-050607-generic) and have Pylon 5.2 installed.

I can see and open all cameras in pylon Viewer in continuous streaming mode, but in the message log I occasionally find errors (especially when I have all 3 cameras running).

I am having problems with the camera occasionally with the code below and even in pylon Viewer: The error that I get is: **Error: 3792764924 UX Status: Libusb error: **UNKNOWN****.

`// Grab.cpp
/*
    Note: Before getting started, Basler recommends reading the "Programmer's Guide" topic
    in the pylon C++ API documentation delivered with pylon.
    If you are upgrading to a higher major version of pylon, Basler also
    strongly recommends reading the "Migrating from Previous Versions" topic in the pylon C++ API documentation.
    This sample illustrates how to grab and process images using the CInstantCamera class.
    The images are grabbed and processed asynchronously, i.e.,
    while the application is processing a buffer, the acquisition of the next buffer is done
    in parallel.
    The CInstantCamera class uses a pool of buffers to retrieve image data
    from the camera device. Once a buffer is filled and ready,
    the buffer can be retrieved from the camera object for processing. The buffer
    and additional image data are collected in a grab result. The grab result is
    held by a smart pointer after retrieval. The buffer is automatically reused
    when explicitly released or when the smart pointer object is destroyed.
*/
// Include files to use the pylon API.
#include <pylon/PylonIncludes.h>
#ifdef PYLON_WIN_BUILD
#    include <pylon/PylonGUI.h>
#endif

#include <opencv2/core/core.hpp>
#include <opencv2/highgui/highgui.hpp>
#include <opencv2/video/video.hpp>

// Namespace for using pylon objects.
using namespace Pylon;
// Namespace for using OpenCV objects.
using namespace cv;
// Namespace for using cout.
using namespace std;
// Number of images to be grabbed.
static const uint32_t c_countOfImagesToGrab = 10000;
int main(int argc, char* argv[])
{
    // The exit code of the sample application.
    int exitCode = 0;
    // Before using any pylon methods, the pylon runtime must be initialized. 
    PylonAutoInitTerm autoInitTerm;
    try
    {
        CTlFactory& TlFactory = CTlFactory::GetInstance();
        CDeviceInfo left_cam;
        left_cam.SetSerialNumber("22822551");
        // Create an instant camera object with the camera device found first.
        CInstantCamera camera( TlFactory.CreateFirstDevice(left_cam));
        // Print the model name of the camera.
        cout << "Using device " << camera.GetDeviceInfo().GetModelName() << endl;
        
        // Get camera nodemap in order to access camera parameters.
        GenApi::INodeMap& nodemap = camera.GetNodeMap();
        // Open the camera before accessing any parameters. 
        camera.Open();
        // Create pointers to access camera width and height
        GenApi::CIntegerPtr width=nodemap.GetNode("Width");
        GenApi::CIntegerPtr height=nodemap.GetNode("Height");
        // The parameter MaxNumBuffer can be used to control the count of buffers
        // allocated for grabbing. The default value of this parameter is 10.
        camera.MaxNumBuffer = 5;
        // Create a pylon ImageFormatConverter object.
        CImageFormatConverter formatConverter;
        // Specify the output pixel format
        formatConverter.OutputPixelFormat=PixelType_BGR8packed;
        // Create a PylonImage that will be used to create OpenCV images later.
        CPylonImage pylonImage;
        // Create an OpenCV image.
        Mat openCvImage;

        // Start the grabbing of c_countOfImagesToGrab images.
        // The camera device is parameterized with a default configuration which
        // sets up free-running continuous acquisition.
        camera.StartGrabbing();
        // This smart pointer will receive the grab result data.
        CGrabResultPtr ptrGrabResult;
        // Camera.StopGrabbing() is called automatically by the RetrieveResult() method
        // when c_countOfImagesToGrab images have been retrieved.
        while ( camera.IsGrabbing())
        {
            // Wait for an image and then retrieve it. A timeout of 5000 ms is used.
            camera.RetrieveResult( 5000, ptrGrabResult, TimeoutHandling_ThrowException);
            // Image grabbed successfully?
            if (ptrGrabResult->GrabSucceeded())
            {
                Access the image data.
                cout << "SizeX: " << ptrGrabResult->GetWidth() << endl;
                cout << "SizeY: " << ptrGrabResult->GetHeight() << endl;
                const uint8_t *pImageBuffer = (uint8_t *) ptrGrabResult->GetBuffer();
                cout << "Gray value of first pixel: " << (uint32_t) pImageBuffer[0] << endl << endl;

                // Convert the grabbed buffer to a pylon image.
                formatConverter.Convert(pylonImage, ptrGrabResult);

                // Create an OpenCV image from a pylon image.
                openCvImage = cv::Mat(ptrGrabResult->GetHeight(), ptrGrabResult->GetWidth(), CV_8UC3,(uint8_t *) pylonImage.GetBuffer());

                // Create an OpenCV display window.
                namedWindow("OpenCV Display Window", CV_WINDOW_NORMAL);
                resizeWindow("OpenCV Display Window", 2048/2,1536/2);

                // Display the current image in the OpenCV display window.
                imshow("OpenCV Display Window", openCvImage);
                waitKey(1);
            }
            else
            {
                cout << "Error: " << ptrGrabResult->GetErrorCode() << " " << ptrGrabResult->GetErrorDescription() << endl;
            }
        }
    }
    catch (const GenericException &e)
    {
        // Error handling.
        cerr << "An exception occurred." << endl
        << e.GetDescription() << endl;
        exitCode = 1;
    }
    // Comment the following two lines to disable waiting on exit.
    cerr << endl << "Press enter to exit." << endl;
    while( cin.get() != '\n');
    // // Releases all pylon resources. 
    // PylonTerminate();  
    return exitCode;
}`

Hope someone can share your expertise based on previous experience based on this error.

thiesmoeller, 2020-07-10:
@Neel1302 , you are running out of USB memory.

You can extend the amount of RAM allocated to the usb subsystem:

In the full pylon readme, there is a section about USB tuning:
```
* Increasing Packet Size
  For faster USB transfers, you should increase the packet size. If you're
  working with the pylon Viewer, you can do this by changing the
  "Stream Parameters" -> "Maximum Transfer Size" value from inside the
  pylon Viewer. If you're writing your own application, use the camera API
  to set the "Maximum Transfer Size".
  After increasing the package size, you will likely run out of kernel space
  and see corresponding error messages in the console. The default value set
  by the kernel is 16 MB. To change the value (in this example to 1000 MB), you can
  execute as root:
    echo 1000 > /sys/module/usbcore/parameters/usbfs_memory_mb
  This would assign a maximum of 1000 MB to the USB stack.

```

Neel1302, 2020-07-10:
Hi @thiesmoeller,

Thanks for that. Yesterday, I did troubleshoot the issue and confirmed that the Maximum Transfer Size under Stream Parameter settings in Pylon Viewer is set to 65535 (bytes) by default. I am not sure if this is a linux xHCI drive related issue, but I on one of the cameras, the default value of 65535 worked ok. However, for the other two cameras, I have to actually reduce the value to around 5000 to get it to transfer images at expected bandwidth of several MB/sec. With the default values, these cameras were just transferring the data at <1MB/sec when I checked with wireshark and usbtop (Pylon seems to report the bandwidth incorrectly compared to these other tools).

This is counter intuitive, but correct me if I am wrong: it seems that lowering the max transfer size encourages frequent but smaller-sized packet transfers/higher bandwidth on the USB lines while increasing this encourages slower but larger-sized packet transfers. This could be the reason why higher kernel space assignment is required - as you pointed in the readme. I noticed some spikes in the packet size as well and it seems that with low bandwidth, the MaxBufferSize may not be sufficient as well if I leave the MaxTransferSize parameter to be at the default value.

If there is a rule of thumb for having a balanced tradeoff between the rate at which packets are sent and the packet size, let me know.

Thanks again for the guidance,
Neel

thiesmoeller, 2020-07-11:
The MaxTransferSize defines the size of the DMA transfer length for the XHCI controller on your host ( [URB](https://www.kernel.org/doc/html/latest/driver-api/usb/URB.html)).
The larger this value is, the lower the CPU overhead, as each transfer involves the CPU handling the fresh data and pushing new memory descriptors to the XHCI.
The packet size on USB3 is always 1024byte. ( or less if the packet can't be filled completely at the end of the image, or for leader/trailer packets )

To have maximum performance you should increase this value ( typically 1MByte gives reasonable performance on most systems )

Are all USB3 ports connected to the same XHCI on your AMD system?
As I don't know your AMD setup...some Intel insights: 
On modern Intel system the root-hub (which is behind the XHCI and handles multiple ports on one XHCI) is not limited to the 5Gbps, that you would have on an external USB3 hub, but can handle 5Gbps on _each_ port without any issues.

To find the optimal settings, for your system, use the pylon bandwidth manager.

[image]

I recommend, that you install the latest pylon version on your system, which includes pylonviewer and the included bandwidth manager. The values found there can than be manually set in your python script.

Neel1302, 2020-07-16:
Thanks again @thiesmoeller,

I have been doing some tests lately. My setup is using a Gigabyte X399 based motherboard [https://www.gigabyte.com/ca/Motherboard/X399-DESIGNARE-EX-rev-10#kf](url) with an AMD Ryzen 2950X CPU. The ports I have connected the cameras to are all USB3.1 ports.

I currently have 3 cameras running and plugged directly to the ports on the motherboard.

With the 4 cameras plugged in, and this is what lsusb shows:
[image].

Here is what I am not able to understand/explain:
With all other cameras disconnected, if I plug in one camera only (USB Bus 6 device ID 3), this is what usbtop shows me for the transfer rate (~106 KiB/s = 0.108544 MB/s):
[image]

Without removing or changing anything, I double check this with wireshark and it is indeed the same (note wireshark plot y axis is in bytes and it adds up the "From Device" and "To Device" fields that you see in usbtop, thus its magnitude seems very close to what we get if we add both the "From Device" and "To Device" fields in usbtop):
[image]

Now in Pylon, with this exact same setup, the bandwidth manager tell me a complete different story (94.4 MB/s):
[image]

Do you know what is going on here as it seems pylon doesn't give me the correct bandwidth. Also the fact that I keep getting libusb error confirm the bandwidth should be low. Surprisingly if I change the Max Transfer Size from default (65536) to 5000 (bytes) the bandwidth increases:
[image]

Yet Pylon reports same bandwidth:
[image]

Thanks,
Neel

thiesmoeller, 2020-07-20:
... to your bandwidth measurements:
If you get 30 full 2048*1536 frames in PixelType_BGR8packed, you transfer 94371840 Byte per second.

all the external tools you use, rely on the usbmon/libpcap infrastructure. So if there is an issue on the system you use, or the software version you use, it is possible, that they all show funny data.

If usbmon/libpcap is not correctly handling the large transfer sizes. This error in MB/s will get large the larger the MaxTransferSize is.

Neel1302, 2020-07-20:
Hi @thiesmoeller,

Thanks once again for the continued support.

I understand your point. Would you suggest I go by pylon Viewer for a rather accurate/reliable bandwidth measurement compared to the others?

Also, I have tried increasing the maxTransferSize to the max value: 4194304 bytes (~4 Mb), however the error: UX Status: Libusb error: **UNKNOWN* still remains. I have modified the usbfs limit to 1000Mb as per your direction pointing towards the manual.

What I have noticed is that usually with one camera it seldom occurs, but with all three cameras running it pops up on the screen quite frequently and results in this exception: 

[image]

The corresponding dmesg output at this time is:
[image]

I have to say that with pylonviewer I do not see this error and it seems like it is happening with the C++ code that I run. I am using the following C++ code:

Neel1302, 2020-09-20:
Hi @thiesmoeller,

I am posting for other users that might run into this issue. The root cause of the issue was that the usb bus bandwidth was a limiting factor. It is not clear whether it was the driver for the PCIe card we were using, but we have since used the [Fresco FL1100, 4HC, x4, 4Ports - USB Interface Card](https://www.baslerweb.com/en/products/vision-components/pc-cards/usb-3-0-interface-card-pcie-fresco-fl1100-4hc-x4-4ports/) from basler and the problem has since been resolved without any changes made to software or camera settings.

Thanks,
Neel

closed 2020-09-20 by Neel1302