basler/pypylon #254: LIBUSB_ERROR_TIMEOUT issue

crunchyluke, 2020-07-14:
**Full error message:** 

"PrepareGrab failed for device '2676:ba03:2:1:2'. Error: 'UX Status: Libusb error: LIBUSB_ERROR_TIMEOUT.' : RuntimeException thrown (file 'PylonUsbStream.cpp', line 260)."

**Platform details:**

I receive this error after several hours of normal operation on a Raspberry Pi 4 with a daA2500-14um ("Dart") camera connected via a USB 3.0 cable to one of the USB 3.0 ports. The same intermittent issue occurs with the same application on an identical RPi4 and connected daA2500-14um pair.

**What I've tried so far to resolve:**

1. Write 1000 to the /sys/module/usbcore/parameters/usbfs_memory_mb file at startup to assign a maximum of 1000 MB to the USB stack.

2. Changed DeviceLinkThroughputLimitMode to 'Off':
camera.GetNodeMap().GetNode('DeviceLinkThroughputLimitMode').SetValue('Off')

3. Set the MaxTransferSize parameter to 4 mb:
camera.GetStreamGrabberNodeMap().GetNode("MaxTransferSize").SetValue(4000000)  # 4 mb MTS

4. Set the MaxNumBuffer parameter to 20:
camera.MaxNumBuffer.SetValue(20)

**Possibly relevant implementation details:**

I use the USB camera in two different threads within my application (one for a constant mjpeg stream, the other for uncompressed png single images). To manage the camera object resource, I use a Lock object (from the Python threading library) to ensure that the camera is not used by both threads at the same time.  Both of these threads get an image from the camera by first acquiring the Lock object, using the StartGrabbing(pylon.GrabStrategy_LatestImageOnly) call to start grabbing and camera.RetrieveResult(1500) to get the grabResult, which is then checked with a grabResult.GrabSucceeded() call. The valid grab result is then encoded into a jpeg and written to a stream (for the mjpeg stream thread) or saved to disk as a png (for the single, uncompressed image thread).

Any suggestions to help solve this issue would be very much appreciated!

stefanklug, 2020-08-25:
This sounds more like a low level issue on the RPi. Are there any messages in dmesg? Can you recover by opening/closing the camera or is the usb connection dead?

crunchyluke, 2020-08-27:
Thanks for the reply. I'll try to reproduce the issue soon then check dmesg and use Pylon Viewer to attempt to open the camera again.

keven425, 2021-06-24:
@crunchyluke any luck resolving this issue? I'm having the same issue.

Just filed one here: https://github.com/basler/pypylon/issues/381

thiesmoeller, 2025-12-15:
This issue describes USB timeout errors after extended operation, which are typically related to USB hardware, drivers, or system-level issues rather than a pypylon bug. The user has already tried several workarounds. We are closing this issue. If you believe this is a pypylon specific bug, please open a new issue with detailed reproduction steps and system logs.