basler/pypylon #255: assemble an image from several lines like a scanner

kernelport, 2020-07-15:
I would like to assemble an image from several lines.
My solution works but requires that the line images are saved temporarily as a file.
Is there a better solution? 
```
"""
Cam:            acA2000-50gc
Pixel Format:   BayerGR12Packed
AOI Controls:   Width = 2046
                Height = 2
"""
from pypylon import pylon
import cv2

num_img_to_save = 1000
img = pylon.PylonImage()
tlf = pylon.TlFactory.GetInstance()

cam = pylon.InstantCamera(tlf.CreateFirstDevice())
cam.Open()
cam.StartGrabbing()

im_v = None
for i in range(num_img_to_save):
    with cam.RetrieveResult(2000) as result:

        img.AttachGrabResultBuffer(result)
        img.Save(pylon.ImageFileFormat_Png, "line.png")
        lines = cv2.imread('line.png')
        height, width = lines.shape[:2]
        line = lines[0:1, 0:width]

        if im_v is None:
            im_v = line
        else:
            im_v = cv2.vconcat([im_v, line])

cv2.imwrite('opencv_vconcat.png', im_v)
cam.StopGrabbing()
cam.Close()
```

thiesmoeller, 2020-07-17:
TODO:
 if saving image takes too long you might loose scanlines
 -> increase the number of buffers to a much higher value to compensate for this.
```
from pypylon import pylon
import numpy as np
import cv2

SCANLINE_HEIGHT = 2
VIRTUAL_FRAME_HEIGHT = 1000

tlf = pylon.TlFactory.GetInstance()

cam = pylon.InstantCamera(tlf.CreateFirstDevice())
cam.Open()

# setup center scan line
cam.Height = SCANLINE_HEIGHT
cam.Width = cam.Width.Max
cam.CenterX = True
cam.CenterY = True

# setup for
cam.PixelFormat = "BayerRG8"
cam.Gain = 34
cam.ExposureTime = 200
print("Resulting framerate:", cam.ResultingFrameRate.Value)
cam.StartGrabbing()

img = np.zeros((VIRTUAL_FRAME_HEIGHT, cam.Width.Value), dtype=np.uint8)
missing_line = np.zeros((SCANLINE_HEIGHT, cam.Width.Value), dtype=np.uint8)
image_idx = 0
while True:

    for idx in range(VIRTUAL_FRAME_HEIGHT // SCANLINE_HEIGHT):
        with cam.RetrieveResult(2000) as result:
            if result.GrabSucceeded():
                img[idx * SCANLINE_HEIGHT:idx * SCANLINE_HEIGHT + SCANLINE_HEIGHT] = result.Array
            else:
                img[idx * SCANLINE_HEIGHT:idx * SCANLINE_HEIGHT + SCANLINE_HEIGHT] = missing_line

    img_rgb = cv2.cvtColor(img, cv2.COLOR_BayerRG2RGB)

    # Display the resulting frame
    cv2.imshow('Linescan View', img_rgb)
    # Save the resulting frame
    cv2.imwrite("sample_linescan%05d.jpg" % image_idx, img_rgb)
    image_idx += 1
    if cv2.waitKey(1) & 0xFF == ord('q'):
        break

# When everything done, release the capture
cam.StopGrabbing()
cv2.destroyAllWindows()

cam.Close()

```

stefanklug, 2020-08-25:
@kernelport I'm closing this, as there was no activity in the last month. If needed please comment & reopen.

closed 2020-08-25 by stefanklug