basler/pypylon #256: InstantCamera.NumQueuedBuffers() reports always a full buffer

SAndreas23, 2020-07-20:
Hi everyone,
I'm running in an issue when using the pypylon package with three Basler ac640-120 gc cameras:
Using the InstantCamera class and the provided GrabLoop-Thread for each instance, i'm trying to monitor the current buffer usage. I'm trying to achieve this by printing InstantCamera.NumEmptyBuffers() in a modified ImageEventHandler in the OnImageGrabbed function. I ran a few tests and recorded the acquired samples using open cv. The number of images and the timespan needed to acquire them matched the defined samplerate, so aparantly no frame drops. Problem is, that NumEmptyBuffers() always reported zero empty buffers. I changed the value of MaxNumBuffers() a few times (10, 20, 50), but the problem stayed the same. I also tried NumEmptyBuffers / NumReadyBuffers() etc. with the same result. The instances of InstantCamera are created uying the TIFactory and are configured to work with a hardwaretrigger, which results in a samplerate of 100 Hz. Am i missing a relevant action to use NumQueuedBuffers() succesfully or is there maybe another issue?

Thanks in advance!

Attached below is a part of my code (which hopefully includes all revlevant information):
The changed ImageEventHandler:

```
class HWTriggerImageEventHandler(pylon.ImageEventHandler):

    def __init__(self, videoname, samplerate):
        self.samplerate = samplerate
        self.videorecorder = cv2.VideoWriter(videoname,
                                             cv2.VideoWriter_fourcc(
                                                 *'mp4v'),   samplerate,
                                             (658, 492))
        # self.videorecorder = cv2.VideoWriter(videoname,
        #                                      cv2.VideoWriter_fourcc(
        #                                          *'XVID'),   samplerate,
        #                                      (658, 492))
        self.converter = pylon.ImageFormatConverter()
        self.converter.OutputPixelFormat = pylon.PixelType_BGR8packed
        self.converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned
        super(HWTriggerImageEventHandler, self).__init__()

    def OnImageGrabbed(self, camera, grabResult):
        image = self.converter.Convert(grabResult).GetArray()
        self.videorecorder.write(image)
        grabResult.Release()
        print('NumQueuedBuffers', camera.NumQueuedBuffers())
```

code where the ImageEventHandler is registered and the acquisistion is started:

```
    def record_cameras(self, cameras, image_count, samplerate, videonames):

        for i, camera in enumerate(cameras):
            try:
                camera.RegisterImageEventHandler(HWTriggerImageEventHandler(
                    videonames[i], samplerate), pylon.RegistrationMode_Append,
                        pylon.Cleanup_Delete)
                camera.MaxNumBuffer = 1000
                camera.StartGrabbingMax(image_count,
                                        pylon.GrabStrategy_OneByOne,
                                        pylon.GrabLoop_ProvidedByInstantCamera)

                print('Grabbing Started')

            except genicam.GenericException as e:
                # Error handling
                print("An exception occurred.", e.GetDescription())
                exitCode = 1
        while True:
            try:
                time.sleep(1)
            except KeyboardInterrupt:
                break

        camera.Close()
```

thiesmoeller, 2020-07-20:
Your image save handler is fast enough.. be happy :-)

To demonstrate the failing case, just sleep a while in the image handler to simulate a toooo slow OnImage callback

This looks looks like this:

```
Grabbing Started
NumQueuedBuffers 22
NumQueuedBuffers 19
NumQueuedBuffers 17
NumQueuedBuffers 14
NumQueuedBuffers 12
NumQueuedBuffers 10
NumQueuedBuffers 6
NumQueuedBuffers 3
NumQueuedBuffers 0
NumQueuedBuffers 0
NumQueuedBuffers 0
NumQueuedBuffers 0
```

This is the test code although number might be different depending on your timing.

```python
import time
from pypylon import pylon

tlf = pylon.TlFactory.GetInstance()

cam = pylon.InstantCamera(tlf.CreateFirstDevice())
cam.Open()

class NumQueuedBufferEventHandler(pylon.ImageEventHandler):
    def __init__(self):
        super(NumQueuedBufferEventHandler, self).__init__()

    def OnImageGrabbed(self, camera, grabResult):
        time.sleep(0.2)
        print('NumQueuedBuffers', camera.NumQueuedBuffers())

num_queued_buffer_handler = NumQueuedBufferEventHandler()

cam.RegisterImageEventHandler(num_queued_buffer_handler, pylon.RegistrationMode_Append,
                        pylon.Cleanup_Delete)
cam.MaxNumBuffer = 25
cam.StartGrabbingMax(30,pylon.GrabStrategy_OneByOne, pylon.GrabLoop_ProvidedByInstantCamera)
print('Grabbing Started')

while(True):
    pass

```

SAndreas23, 2020-07-21:
Thanks for the reply.
 First of all to clearify my request: I mentioned NumQueuedBuffers() in the headline, which reports (using my code) always the same value as MaxNumBuffer(). NumEmptyBuffers() always reports zero as i mentioned. 
So now i'm a little bit confused. If i understand your answer correctly this means, that if the reported value by NumQueuedBuffers() is zero, there are no available buffers anymore and i am likely to loose frames?

Thanks in advance!

thiesmoeller, 2020-07-21:
> that if the reported value by NumQueuedBuffers() is zero, there are no available buffers anymore and i am likely to loose frames?

Yes the queued buffers represent those buffers, that don't belong to your application code, but to the grab engine. 
And if the grab engine has no queued buffer available, while receiving a new frame, it will skip this frame.

You can check this with the other callback provided:

https://github.com/basler/pypylon/blob/a3ca9123979268ddc6ae986cb35f75f6523f5481/samples/imageeventprinter.py#L7

SAndreas23, 2020-07-22:
Allright now i get it! Thanks very much!

closed 2020-07-22 by SAndreas23