basler/pypylon #257: Set ExposureTime for Multiple Cameras

HajimeMiyata, 2020-07-21:
I want to change ExposureTime for each cameras to each ExposureTime.

I make python program attached program.
However, the following error occurs.

I guessed I mistake the way to access to CInstantCameraArray. 
Do you know how to access each camera's parameters?
=============================
An exception occurred.
Traceback (most recent call last):
  File ".\grab_ExposureTime_Mulpiple_cameras.py", line 66, in <module>
    camera[0].ExposureTiSetValue(3000.0)
  File "C:\Users\hmiyata\AppData\Local\Programs\Python\Python37\lib\site-packages\pypylon\pylon.py", line 4039, in __getattr__
    return self.GetNodeMap().GetNode(attribute)
  File "C:\Users\hmiyata\AppData\Local\Programs\Python\Python37\lib\site-packages\pypylon\genicam.py", line 1473, in GetNode
    return _genicam.INodeMap_GetNode(self, Name)
_genicam.LogicalErrorException: Node not existing (file 'genicam_wrap.cpp', line 16599)
[image]
[grab_ExposureTime_Multiple_cameras.txt](https://github.com/basler/pypylon/files/4951138/grab_ExposureTime_Multiple_cameras.txt)

thiesmoeller, 2020-07-22:
the error message already points to your typo:

`camera[0].ExposureTiSetValue(3000.0)`

which has to be like you correctly did with all other features: [ feature names from [docs.baslerweb.com](https://docs.baslerweb.com/exposure-time) ]

`camera[0].ExposureTime.SetValue(3000.0)`

HajimeMiyata, 2020-07-28:
Dear thiesmoeller,

Thank you for your reply.
It is simple spelling miss.

And, we  found     camera.Open() is needed.
Finally, we can solve it by setting ExposureTime following code.
===============

    camera.Open()
    print("pass")
    #Set ExposureAuto
    #Mode:Continuous,Once,Off
    camera.ExposureAuto = "Off"

    print("pass1")
    camera[0].ExposureTime.SetValue(3000.0)
    print("pass2")
    Mode = camera[0].ExposureMode.GetValue()
    print("ModeCurrent:",Mode);
    camera[0].ExposureMode.SetValue("Timed")

    Mode = camera[0].ExposureMode.GetValue()
    print("Mode Setting:",Mode);

    camera[1].ExposureTime.SetValue(3000.0)
============================

closed 2020-08-25 by stefanklug