basler/pypylon #259: Does anyone know why Basler cámaras don't run with multiprocessing

zamarjavier, 2020-07-27:
Hi,

I'm using two Basler cameras, concretely 1920um model. I'm trying to capture synchronized captures. One camera is the master camera and the other camera is the slave. The master triggers the salve using a software trigger. 
To achieve this, I use threading python library. I use one thread per camera.
When I developed the code with Python Threading library, the process goes well, but when I migrate the code to Python multiprocessing which is similar, the cameras don't go. It seem that they are not recognized .

Does anyone know what is happening?

Otherwise, does anyone know how to achieve camera triggering in slave/master mode without threads?

Thanks in advance.

mathieu-chauvet, 2020-07-27:
Hi, 

I don't know how synchronous the image capture must be in your application, but a good way to sync grabbing is to use a hardware trigger connected to both cameras (this makes sure you're in sync). 

Then you can have two python threads that are connected to each camera and gather the data. 

I hope it helps

zamarjavier, 2020-07-27:
first of all thank you for your answer. I did as you mention on your comment. 

My problem is if I use Multiprocessing instead of Threading module, the cameras don't run. 

I am considering to use multiprocessing because using threading I have a memory accumulation.

Is it normal to have memory accumulation in a long time duration program?

¿How can I check the memory accumulation?

VipulPandey-arch, 2020-08-31:
> Hi,
>
> I'm using two Basler cameras, concretely 1920um model. I'm trying to capture synchronized captures. One camera is the master camera and the other camera is the slave. The master triggers the [...]

hi ,may i know how you grabbed and stored image in ur PC using pypylon library.

antortjim, 2020-10-23:
Hi @zamarjavier I had the same problem of memory accumulation but turns out it was due to a mistake I made in my Python code, where a function was fetching one frame and calling itself recursively. Every call was storing a new frame in the computer's RAM and within < 1 minute the RAM would get full and the computer would freeze. Check you don't have recursive calls like this or make sure they are not causing this problem. You can visualize the RAM consumption in Linux with the system monitor or just by running `free -h` in the terminal continuously.

thiesmoeller, 2025-12-15:
This issue regarding multiprocessing with Basler cameras has been addressed in the comments. The cameras work with threading but not multiprocessing due to how camera objects are handled. We are closing this issue. For further questions on multiprocessing, please refer to the discussion or open a new issue.