basler/pypylon #260: ImportError: cannot import name '_pylon' from 'pypylon'

aligog, 2020-07-28:
I've installed pylon_6.1.3.20159-deb0_arm64.deb on jetson tx1
and after that installed pypylon-1.5.4-cp35-cp35m-linux_aarch64.whl
but when I test python3 -c "import pypylon.pylon" that exception occur:

```
python3.5 -c "import pypylon.pylon"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/nvidia/.local/lib/python3.5/site-packages/pypylon/pylon.py", line 40, in <module>
    from . import _pylon
ImportError: cannot import name '_pylon'
```

SMA2016a, 2020-07-28:
maybe the pylon has been installed in python2xx environment. 

try this command to see which python version is the default one.
python -c "import sys ; print(sys.platform, sys.path)"

more easier for you if work with virtual enviremnt created with env.

A Virtual environment can be created with the following command:

python3 -m venv /opt/pypylonproject

now enter the virtual environment with the command:

source /opt/pypylonproject/bin/activate

then install pypylon

aligog, 2020-07-29:
> maybe the pylon has been installed in python2xx environment.
>
> try this command to see which python version is the default one.
> python -c "import sys ; print(sys.platform, sys.path)"
>
> more easier for [...]

i don't think related because pypylon was installed on python3 and is in site-packages
and also pylon was installed and is in './opt' path

SMA2016a, 2020-07-29:
would you post the output of following command?

python -c "import sys ; print(sys.platform, sys.path)"

aligog, 2020-07-29:
> would you post the output of following command?
> 
> python -c "import sys ; print(sys.platform, sys.path)"

```
python -c "import sys ; print(sys.platform, sys.path)"
('linux2', ['', '/usr/lib/python2.7', '/usr/lib/python2.7/plat-aarch64-linux-gnu', '/usr/lib/python2.7/lib-tk', '/usr/lib/python2.7/lib-old', '/usr/lib/python2.7/lib-dynload', '/home/nvidia/.local/lib/python2.7/site-packages', '/usr/local/lib/python2.7/dist-packages', '/usr/lib/python2.7/dist-packages', '/usr/lib/python2.7/dist-packages/gtk-2.0'])
```

thiesmoeller, 2020-07-29:
As quick fix please compile from source in your environment

TPU2020, 2020-08-13:
Same issue when I try to test import pylon or genicam
I'm using **Jetson NX** and **Code-OSS** environment
I've succesfully installed **pypylon-1.5.4-cp36-cp36m-linux_aarch64.whl**
and I've install and run **pylon_6.1.3.20159-deb0_arm64.deb**

` python3 -c "import pypylon.pylon"`
`Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/tomaspuda/.local/lib/python3.6/site-packages/pypylon/pylon.py", line 40, in <module>
    from . import _pylon
ImportError: cannot import name '_pylon'`

` python3 -c "import pypylon.genicam"`
`Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/tomaspuda/.local/lib/python3.6/site-packages/pypylon/genicam.py", line 40, in <module>
    from . import _genicam
ImportError: cannot import name '_genicam' `

Any solution for this? Thanks in advance

thiesmoeller, 2020-08-13:
Please compile from source. 
1.5.4 arm 64 binary package propably broken

TPU2020, 2020-08-26:
@stefanklug I did compile from source last time - it worked. So now I've done update via pip3 install .whl, tested code and it worked.
Thanks for help.

stefanklug, 2020-08-28:
@TPU2020 
> Is there command for pypylon version check?

You mean something like
```
from pypylon import pylon
print(pylon.__version__)
```

closed 2020-08-28 by stefanklug

aquilesC, 2020-12-22:
I had the same problem with Anaconda on Ubuntu 20.04. Tried to install pypylon from source, but I faced this issue: #308 

After working around it, and installing pypylon using ``setup.py install``, it seems to work.