basler/pypylon #262: Stop getting image by trigger after change system time to back

awarm, 2020-08-04:
Hi. I have a Basler acA2040-35gm. 
The camera is set to receive an image by a trigger from Line1

```   camera.TriggerSelector.SetValue("FrameStart")
   camera.TriggerMode.SetValue("On")
   camera.TriggerActivation.SetValue("RisingEdge")
   camera.TriggerSource.SetValue('Line1')
   camera.StartGrabbing(pylon.GrabStrategy_LatestImageOnly)
```

If I change the time back by a minute or more, the camera will stop receiving the trigger event
Can anyone confirm this issue?

niallodowd, 2020-10-05:
Hi Awarm, 

do you mind sharing the rest of your code? my camera automatically switches off camera trigger mode when I start grabbing frames. I'd like to see how you call image grabbing, maybe there is a special way to keep hardware triggering on and capturing images.

thiesmoeller, 2025-12-15:
This issue describes a problem with trigger reception after system time changes, which appears to be related to system-level time synchronization rather than a pypylon bug. We are closing this issue. If you have evidence this is a pypylon specific issue, please open a new issue with detailed reproduction steps.