basler/pypylon #264: Import Error pylon not a module of pypylon

SpangeJ, 2020-08-10:
Hello,
I am trying to install pypylon on the [NVIDIA XAVIER NX ](https://developer.nvidia.com/embedded/jetson-xavier-nx-devkit).
It is an aarch64 system.
I am using Python 3.6.9 in a virtual environment and I have successfully installed _pypylon-1.5.4-cp36-cp36m-linux_aarch64.whl_

I can import _pyplon_, but the module pylon does not exist.
Can someone help me sort this out?

`from pypylon import pylon`
yields:
`Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/xaviernx1/projects/fishrecordingarm/venv/lib/python3.6/site-packages/pypylon/pylon.py", line 40, in <module>
    from . import _pylon
ImportError: cannot import name '_pylon'
`

NOTE: `import pypylon` does not fail

TPU2020, 2020-08-13:
Same issue when I try to test import pylon or genicam
I'm using Jetson Xavier NX and Code-OSS environment
I've succesfully installed pypylon-1.5.4-cp36-cp36m-linux_aarch64.whl
and I've install and run pylon_6.1.3.20159-deb0_arm64.deb

python3 -c "import pypylon.pylon"
Traceback (most recent call last): File "<string>", line 1, in <module> File "/home/tomaspuda/.local/lib/python3.6/site-packages/pypylon/pylon.py", line 40, in <module> from . import _pylon ImportError: cannot import name '_pylon'

python3 -c "import pypylon.genicam"
Traceback (most recent call last): File "<string>", line 1, in <module> File "/home/tomaspuda/.local/lib/python3.6/site-packages/pypylon/genicam.py", line 40, in <module> from . import _genicam ImportError: cannot import name '_genicam'

thiesmoeller, 2020-08-13:
Please compile from source
1.5.4 arm64 binary package probably broken

TPU2020, 2020-08-13:
> Please compile from source
> 1.5.4 arm64 binary package probably broken

I'm quite not sure if I understand correctly what do you mean by "compile from source"
Could you describe it a bit more or got any specific example?
Thanks for help again.

thiesmoeller, 2020-08-16:
https://github.com/basler/pypylon/blob/master/README.md

SpangeJ, 2020-08-18:
> Please compile from source
> 1.5.4 arm64 binary package probably broken

`pip install .`
Fails, gives the following error:

`    ERROR: Command errored out with exit status 1:
     command: /home/xaviernx1/projects/fishrecordingarm/venv/bin/python -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-req-build-lpt8d0p1/setup.py'"'"'; __file__='"'"'/tmp/pip-req-build-lpt8d0p1/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-16v4w7t7
         cwd: /tmp/pip-req-build-lpt8d0p1/
    Complete output (11 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-req-build-lpt8d0p1/setup.py", line 804, in <module>
        bs = BuildSupport.make()
      File "/tmp/pip-req-build-lpt8d0p1/setup.py", line 321, in make
        return BuildSupportLinux()
      File "/tmp/pip-req-build-lpt8d0p1/setup.py", line 608, in __init__
        self.SwigExe = self.find_swig()
      File "/tmp/pip-req-build-lpt8d0p1/setup.py", line 125, in find_swig
        raise RuntimeError("swig executable not found on path!")
    RuntimeError: swig executable not found on path!
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
`

thiesmoeller, 2020-08-18:
`sudo apt-get install swig`

SpangeJ, 2020-08-18:
> `sudo apt-get install swig`

Excellent!
Thank you!
The issue has been resolved!

closed 2020-08-18 by SpangeJ