basler/pypylon #265: Opening Multiple Cameras: PrepareGrab failed - request time out

thegyro, 2020-08-14:
I am trying to open 6 cameras (acA5472-17um) attached to a Jetson device through two USB hubs (3 cameras connected on each hub). I am running pypylon inside a docker container running Ubuntu.

I am doing so by:
```
for camera in cameras:
    camera.Open()
    camera.GetStreamGrabberNodeMap().GetNode("MaxTransferSize").SetValue(10 ** 6)
    camera.ExposureTime.SetValue(50 * 1000)
    camera.StartGrabbing(pylon.GrabStrategy_LatestImageOnly)
```

However, when I do this, some cameras open and start grabbing, but others don't I get the following error - 

`_genicam.RuntimeException: PrepareGrab failed for device '2676:ba02:2:4:13'. Error: 'The requested operation timed out.' : RuntimeException thrown (file 'PylonUsbStream.cpp', line 260)`

After this the corresponding camera gets disconnected and I have to replug it for it be recognized by pylon again. Also, this doesn't happen always. Sometimes I get lucky and all cameras open and start grabbing. It would be great if I can get some context on why this is happening and how to go about fixing it.

thiesmoeller, 2020-08-17:
Did you increase the usbfs memory?

thegyro, 2020-08-17:
@thiesmoeller Yes, I did. I tried `echo 1000 > /sys/module/usbcore/parameters/usbfs_memory_mb`.  The problem is, it doesn't happen always. It happens sometimes, but when it does, every subsequent call fails with the same error. If I unplug everything and plug back in, it works for a while (say 5-6 runs and the 7th fails with the above error and all subsequent runs fail too).

thiesmoeller, 2020-08-17:
okay... to further track this down:

* pylon version and pypylon version that you used
* any kernel messages, when this error happens?
* camera resolution and pixelformat, that you use ( Mono12 ? )
* what jetson model? -> memory size

thegyro, 2020-08-17:
- pylon version pylon_6.1.3.20159 and pypylon version 1.5.4+pylon6.1.0 (I installed from source from the pypylon github page)
- I don't see anything when I do `dmesh -wH`
- I am using 5 cameras of model acA5472-17um ( camera resolution obtained via `camera.Width` and `camera.Height` equals `5496x3672` for one of them and `5472x3648` for 4 of them with Pixel Format `Mono8`) and the 6th camera is of model acA2440-35uc (camera resolution is 2448x2048 - with PixelFormat `BayerRG8`)
- Jetson model is AGX Xavier Dev Kit - memory is 32 GB

thegyro, 2020-08-17:
@thiesmoeller I also tried increasing the USB FS memory to 3 GB (through `echo 1000 > /sys/module/usbcore/parameters/usbfs_memory_mb`) - Now I am encountering another problem. My use case is that I want to continuously acquire the images in background but use only the latest one. When I try to `RetrieveResult()` in a loop, after 10-20 seconds, one of the cameras automatically disconnects and I get a `QueueBuffer() failed: Device has been removed from the PC. : RuntimeException thrown` error.  I monitor the RAM and it doesn't look more than 1.5 GB is being utilized. I am not exactly sure why this happens?

thiesmoeller, 2020-08-18:
for your last error you should definitely see kernel trace messages

thegyro, 2020-08-18:
@thiesmoeller Do you mean through something like `dmesg`? I do see a `usb 2-3.2: USB disconnect, device number 95` message and one of the cameras gets disconnected. Do you have any suggestions on how I can better understand what the underlying issue is?

thiesmoeller, 2020-08-20:
From what you describe the issue doesn't seem to be in the pypylon layers ( the python binding ).
Please reach out to your basler support.contact for further assistance in getting your system setup stable.

MattsProjects, 2020-08-20:
Hello,
Short question - are these powered USB hubs, and can you name the brand/model? Also, how long are the USB cables between the cameras & hubs, and the hubs & cameras (and can you name the brand/model there too)?
Many times dropped connections could be a sign of inadequate power/signal strength to the camera. Often this traces back to some substandard accessories.
PS: You could also try disabling the Autosuspend feature in Linux if you haven't already. Sometimes Linux likes to disable USB devices when they are inactive for some time, but the criteria of "activity" is not so well know.
Instructions for this are in an application note found here:
[https://www.baslerweb.com/fp-1507807893/media/downloads/documents/application_notes/AW00145002000_AppNote_Using_Single_Board_Computers_with_Basler_USB_and_GigE_Cameras.pdf](https://www.baslerweb.com/fp-1507807893/media/downloads/documents/application_notes/AW00145002000_AppNote_Using_Single_Board_Computers_with_Basler_USB_and_GigE_Cameras.pdf)

thiesmoeller, 2025-12-15:
This issue about opening multiple cameras has been addressed with guidance on USB hub limitations and camera initialization. We are closing this issue. If you continue to experience issues with multiple cameras, please open a new issue with your specific setup details.

closed 2025-12-15 by thiesmoeller