basler/pypylon #266: Can't Grab image

huwade, 2020-08-20:
Hi, every

I have a acA2440-20gm camera, I use anaconda, python version is 3.7, in win10. 
I following the Hello World sample code, I can get camera.Width.GetValue() but grabResult.GrabSucceeded() is False. 

However I used pylon Viewer 64-bit I can connect to the camera and take a shot. So I don't understand why grabResult.GrabSucceeded() is False.

I would be thankful for any help.

VipulPandey-arch, 2020-08-31:
hello, i am also working on similar project,with pylon viewer u can use pypylon library to grab image..and yes in my coding camera is detected but i am not able to grab image or take a shot.

huwade, 2020-09-01:
Hi, 
I found my buffer is not bigger enough, because I use Network interface controller to grab image. After I change grab size I can take a shot.

thiesmoeller, 2020-09-01:
Please check if the camera works on your system if you use pylon viewer first

thiesmoeller, 2020-09-04:
Can you run the first example referenced in the readme ?https://github.com/basler/pypylon/blob/master/samples/grab.py

This contains the code to output the error description on failure ....

thiesmoeller, 2020-09-08:
Hi @huwade and @VipulPandey-arch,

can you run pylon viewer and as next step: please run the above mentioned grab sample

huwade, 2020-09-14:
Yes, I can run.

VipulPandey-arch, 2020-09-14:
@huwade @thiesmoeller  
 Thanks for the support,
 We have successfully grabbed and saved images by single shot,now we want to grab the image of moving object with software 
 trigger to camera. please find below code and suggest the modification to be done to grab continuously moving object.
P.S i am using Basler Line Scan camera ral 8192-12gm.

import numpy as np
import cv2
from datetime import datetime

from pypylon import pylon

camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
camera.Open()
# # datetime object containing current date and time
now = datetime.now()
 
print("now =", now)

# dd/mm/YY H:M:S
dt_string = now.strftime("%d%m%Y%H%M%S")
vip = int(dt_string)

# demonstrate some feature access
new_width = camera.Width.GetValue() - camera.Width.GetInc()
if new_width >= camera.Width.GetMin():
    camera.Width.SetValue(4000)
    camera.Height.SetValue(7000)

   

    numberOfImagesToGrab = 1
    camera.StartGrabbingMax(numberOfImagesToGrab)

    while camera.IsGrabbing():    
        grabResult = camera.RetrieveResult(10000, pylon.TimeoutHandling_ThrowException)
   
        if grabResult.GrabSucceeded():
            print('hiiii')
        # Access the image data.
            print("SizeX: ", grabResult.Width)
            print("SizeY: ", grabResult.Height)
            img = grabResult.Array
            print("Gray value of first pixel: ", img[0, 0])
       

            filename = 'Image_'
            img = cv2.imwrite(f'D:\Print\{filename + str(vip)}.jpg', img)
            print("Image saved!")
            grabResult.Release()
   
   
camera.Close()
camera.StopGrabbing()
cv2.destroyAllWindows()

MCilento93, 2020-12-22:
> Hi, 
> I found my buffer is not bigger enough, because I use Network interface controller to grab image. After I change grab size I can take a shot. 

What do you mean by network interface and grab size?

closed 2020-12-23 by huwade

MCilento93, 2020-12-27:
> Hi,
> I found my buffer is not bigger enough, because I use Network interface controller to grab image. After I change grab size I can take a shot.

Because I have the same issue.
In pypylon I use `numberOfImagesToGrab = 100` now.
From Pylon software kit everything is ok

Bonitodelcapo, 2023-11-22:
Hi @thiesmoeller i get the following error when executing the example from the README:

Error:  3774873620 The buffer was incompletely grabbed. This can be caused by performance problems of the network hardware used, i.e. network adapter, switch, or ethernet cable. To fix this, try increasing the camera's Inter-Packet Delay in the Transport Layer category to reduce the required bandwidth, and adjust the camera's Packet Size setting to the highest supported frame size.

How can i modify the abovementioned Inter-Packet Delay and Packet Size ?

thiesmoeller, 2023-11-22:
Parameters are described here: https://docs.baslerweb.com/network-related-parameters
But we recommend to first use Basler pylon viewer to setup everything that the camera works.
In the Tools Menu you have the https://docs.baslerweb.com/bandwidth-manager
And in pylon 7.4.0 you have the PylonGigEConfigurator Tool that optmizes your interface settings https://docs.baslerweb.com/overview-of-the-pylon-gige-configurator

So the typical workflow is to get the camera running in pylon viewer first and then move to pypylon.