basler/pypylon #267: BCON for MIPI Camera: no device found

chillking2, 2020-08-24:
I'm trying to use a camera with BCON for MIPI interface on a DART-MX8M-MINI, but don't get it to work.
I have the same issue as #206 , but when trying to `python3 setup.py develop` I get either this error:

`/home/root/pypylon-master/generated/genicam_wrap.cpp:205:11: fatal error: Python.h: No such file or directory
 # include <Python.h>
           ^~~~~~~~~~
compilation terminated.
error: command 'aarch64-poky-linux-gcc' failed with exit status 1`

or this one:
`/home/root/pypylon-master/generated/genicam_wrap.cpp:46924:1: fatal error: error writing to -: Broken pipe
 }
 ^
compilation terminated.
error: command 'aarch64-poky-linux-gcc' failed with exit status 1
`

I followed the Wiki pages for Yocto configurations. 
I added the following to my local.conf:
`IMAGE_INSTALL_append = " \
        swig \
        gcc \
        tcf-agent \
        openssh-sftp-server \
        python3-dev \
        opencv \
        basler-camera-driver \
        pylon \
        python3-pypylon \
        gentl-producer \
        xauth \
"`
And in setup.py I change in PylonConfig "pylon5" to "pylon6". 

Does anyone have a tipp for me how to proceed?
Cheers, chill

thiesmoeller, 2020-08-25:
You report an error message while compiling pypylon.

By adding ` IMAGE_INSTALL_append = "... basler-camera-driver \ pylon \ python3-pypylon \ gentl-producer " `

you already installed pypylon to your image. 

Have you followed the installation process for https://www.baslerweb.com/en/sales-support/downloads/software-downloads/basler-camera-enablement-package/ ?

chillking2, 2020-08-25:
I understood `python3 setup.py develop` is used to ""link" the local pypylon source directory into your python installation". 

So far I used this tutorial: https://variwiki.com/index.php?title=MX8_Basler_Camera

thiesmoeller, 2020-08-25:
When you follow the varscite wiki you have an system image, that has all components of pylon fully installed.

the python3-pypylon package is installed,
And you can just use it with the python3 interpreter on the system.

No further installation is required

chillking2, 2020-08-25:
Thanks for your answer. So I guess it was correct what I did. 
But when I try to communicate with the camera I'm getting an error saying there is no device available:

```
root@imx8mm-var-dart:~# python3
Python 3.5.5 (default, Aug 18 2020, 14:01:32)
[GCC 7.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from pypylon import pylon
>>> pylon.TlFactory.GetInstance().EnumerateDevices()
()
>>> pylon.TlFactory.GetInstance().CreateFirstDevice()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3.5/site-packages/pypylon/pylon.py", line 1708, in CreateFirstDevice
    return _pylon.TlFactory_CreateFirstDevice(self, *args)
_genicam.RuntimeException: No device is available or no device contains the provided device info properties. : RuntimeException thrown (file 'TlFactory.cpp', line 694)
```

Do you know what I have to do then?

thiesmoeller, 2020-08-25:
most probably there is an environment variable missing, that points to the location of the userland portion of the driver-stack

`export GENICAM_GENTL64_PATH=/opt/dart-bcon-mipi/lib`

or add it to /etc/environment:

```
root@imx8mevk:~# cat /etc/environment 
#
# This file is parsed by pam_env module
#
# Syntax: simple "KEY=VAL" pairs on separate lines
#
GENICAM_GENTL64_PATH=/opt/dart-bcon-mipi/lib

```

chillking2, 2020-08-25:
That worked vor me, thank you very much!

closed 2020-08-25 by chillking2