basler/pypylon #268: Synchronize 2 Basler ac1920 camera

ehsanmqn, 2020-08-26:
Helo,
I am working with a Basler ac1920-40gm and a ac1920-40gc cameras together for about 3 month. Nowadays I have some problem around synchronizing these cameras together. I, checked out documentations and read about PTP also. After that I could synchronize these two cameras together at starting point, but, unfortunately they will not synchronized anymore 4 to 12 hours later. If anybody guide me, I will be appriciate.

Frame rate: 15
My code to initiate cameras:
`camera3 = pylon.InstantCamera(tlFactory.CreateDevice(cameras[CAM3_ID]))
        camera = pylon.InstantCamera(tlFactory.CreateDevice(cameras[CAM1_ID]))
        camera3.StartGrabbing(pylon.GrabStrategy_LatestImages)
        camera.StartGrabbing(pylon.GrabStrategy_LatestImages)

        camera.MaxNumBuffer = CAM1_BUFF_SIZE
        camera.GainAuto.SetValue(CAM1_GAIN)
        camera.ExposureAuto.SetValue(CAM1_SHUTTER)
        camera.AcquisitionFrameRateAbs.SetValue(CAM1_FRAME_RATE)
        camera.GainRaw.Value = CAM1_INIT_GAIN
        camera.ExposureTimeRaw.Value = CAM1_INIT_SHUTTER
        camera.GevIEEE1588.SetValue(True)

        camera3.MaxNumBuffer = CAM3_BUFF_SIZE
        camera3.GainAuto.SetValue(CAM3_GAIN)
        camera3.ExposureAuto.SetValue(CAM3_SHUTTER)
        camera3.AcquisitionFrameRateAbs.SetValue(CAM3_FRAME_RATE)
        camera3.GevIEEE1588.SetValue(True)`

thiesmoeller, 2020-09-02:
Your code:
You have to call startgrabbing _after_ your other initializations.

Hints to trace issue:
To get deeper insight what is happening in your setup 
you could run Wireshark in capture file rotating mode and capture from monitor port of a managed switch.
If you capture everything except gvsp it should be sufficient.

Stop the wireshark trace if your application detects, that the cameras lost synchronization.

thiesmoeller, 2020-12-01:
@ehsanmqn any feedback?

ehsanmqn, 2021-02-17:
Dear @thiesmoeller,
I have started grabbing after initialization as recommended. I also captured packets using Wireshark, but I didn't detect any synchronization problem. I also print offset from Master, in Slave camera and it is tolerating around 0 that means PTP synchronization is working perfectly. Again, after some hours, I see the gc camera give frames behind gm camera. 

I tried to solve this problem by restarting cameras every 6 hour. This take 1 second from me, but if, it was a better solution for this problem I have appreciated that.

thiesmoeller, 2025-12-15:
This issue about camera synchronization requires more information about the specific synchronization requirements and current setup. Since we haven't received updates, we are closing this issue. If you are still experiencing synchronization problems, please open a new issue with detailed information about your synchronization needs.