basler/pypylon #269: Unsteady frame rate when recording with two cameras

wweertman, 2020-08-28:
Hey there! I am trying to record from two cameras at once. I have been able to record from both at once and get it to work. However, I have been unable to stabilize my frame rate. The cameras are not hardware synced, this is okay, they are close enough for our purposes. However, the frame rate is all over the place. 

[image]

my script is pretty long sorry about that. Basically, I stream from both cameras and watch one, following a keyboard interrupt exception error the script begins to capture to .tiff until another keyboard interrupt. After that, I compress the images using PIL and follow that by getting rid of the raw images. I want a frame rate of around 50FPS for each camera and was unable to compress the images inline with the streaming to achieve that. 

Now the one thing I'm not sure of is whether the recording jiggle I'm seeing is real or an artifact of how I am tracking time in the script using the time library. I want to say it is real. 

Cheers!

# -*- coding: utf-8 -*-
"""
Created on Thu Aug 27 17:32:37 2020

@author: wlwee
"""

# Grab_MultipleCameras.cpp
# ============================================================================
# This sample illustrates how to grab and process images from multiple cameras
# using the CInstantCameraArray class. The CInstantCameraArray class represents
# an array of instant camera objects. It provides almost the same interface
# as the instant camera for grabbing.
# The main purpose of the CInstantCameraArray is to simplify waiting for images and
# camera events of multiple cameras in one thread. This is done by providing a single
# RetrieveResult method for all cameras in the array.
# Alternatively, the grabbing can be started using the internal grab loop threads
# of all cameras in the CInstantCameraArray. The grabbed images can then be processed by one or more
# image event handlers. Please note that this is not shown in this example.
# ============================================================================

import os

os.environ["PYLON_CAMEMU"] = "3"

from pypylon import genicam
from pypylon import pylon
import sys
import time

#enter the trial name here, date time will automatically be appended to the folder name
test_name = 'dual_camera_compression_test'
#enter the folder where you want the trial to be stored, make sure you end with a '/'
folder_path = 'C:/Users/wlwee/Documents/python/fhl_pypylon/DATA/dual_camera_test/'

#camera_sleep_time = 0.0001 #sleep timer with the script, better off adjusting c_fps

# adjusting exposure and gain should be done in the pylon viewer then set here
# Mono8 or Mono12 are the two pixel formats to consider, Mono8 takes less data
c_exposure, c_gain, c_fps = 25000.0, 5.0, 100.0
c_pixel_format = "Mono8"

#replace with False if you do not wish to compress images immediatly after the trial
compress_raw_images = True
#replace with False if you do not wish to replace the raw images after the trial, only if compress_raw_images = 'yes'
remove_raw_images = True

#jpeg quality modification, 100 - i * 10
#0 is the highest quality
i = 0
ipo = pylon.ImagePersistenceOptions()
q = 100 - i * 10
ipo.SetQuality(q)

#raw image codec of .tiff works right now
image_codec = 'tiff'
image_codec_compress = ''

if image_codec == 'tiff':
    #raw image codec of .jpeg works right now and nothing else
[... 376 lines cut ...]
        
        print()
        t_c = time.perf_counter() - t_c
        print('time taken to compress: ' + str(t_c))
        print('compression time/trial time: ' + str(t_c/t_s))
        
        if remove_raw_images == True:
            
            import shutil
            
            print()
            print('removing raw images')
            print()
            
            t = time.perf_counter()
            shutil.rmtree(codec_path)
            print('time taken to delete raw images: ' + str(time.perf_counter() - t))
        
    except KeyboardInterrupt:
        print('keyboard interupt during compression')
        
print()
plt.show()

sys.exit(exitCode)

wweertman, 2020-08-28:
oh my, I did not realize it would format like that. I am so sorry!

wweertman, 2020-08-28:
[image]
Here is another example. Why does the frame rate spike like this?

thiesmoeller, 2020-08-28:
Please edit your original comment to have proper code formatting
You could always use the preview pane before posting.

What is the camera that you are using ?
OS and platform?

wweertman, 2020-08-28:
> Please edit your original comment to have proper code formatting
> You could always use the preview pane before posting.
> 
> What is the camera that you are using ?
> OS and platform?

Windows 10 Pro
Python 3.7.3
processor: Intel(R) Core(TM) i9-9880H
RAM: 32GB
System Type: 64-bit

I have changed the format, it seems more readable. I do know how to get it all as 'code' chunks. It seems like GitHub is forgetting some of my tabs in the early if statements. However, if I open to edit I can see the tabs. This text file contains the code with proper tabs. 
[dual_camera_code_8_27.txt](https://github.com/basler/pypylon/files/5144169/dual_camera_code_8_27.txt)

thiesmoeller, 2020-08-31:
to the editing ... the normal markdown for multiline code sections is triple backticks ( or more than in your code, where you already use single backticks ... )

````
```
   code
   code 
```
````
to the timestamps:

your code only captures the timestamps from the host. You are running it on a non realtime operating system. So the jitter in the millisecond range is expected. 

But the camera timestamps each image captured with a high precision.

To get the timestamp, that is embedded in the image just call after retrieving:

`ts_ns = grabResult.TimeStamp`

This timestamp is a 64bit value in nanoseconds since boot of the camera. 

`ts_s = ts_ns * 1.e-9`

wweertman, 2020-08-31:
[image]

thank you! Beautiful! The cameras are stable.
I replaced the capture timestamp from the host code and now there is no jiggling.

closed 2020-09-01 by thiesmoeller

nchauhan5, 2021-03-19:
@thiesmoeller - How can I print the resulting frame rate coming out from the camera using the python pypylon API? I see some code used here like - cam.ResultingFrameRate.GetValue()

But when I use it, it gives me error like - 
_genicam.LogicalErrorException: Node not existing (file 'genicam_wrap.cpp', line 16130)

Is it removed from the library now? I have pypylon==1.7.2, windows 10, Basler aca4600-7gc Ace GigE camera with 7 frames per second supported.

thiesmoeller, 2021-03-20:
Please check the _model specific_ documentation for your camera at docs.baslerweb.com

cam.ResultingFrameRateAbs.GetValue()

barliya, 2022-01-19:
Hi,

I would like to piggyback on this thread if that's ok.
I am using an acA1440-220um camera and trying to get 30 seconds recording sessions at 200fps.
The expected results would be 6000 frames with timestamps difference of 5 milliseconds.

My code looks something like that:
```
while camera.IsGrabbing():
    # consider if the timeout handler should throw an exception or just ignore.
    grab = camera.RetrieveResult(
        timeout, pylon.TimeoutHandling_ThrowException)
    if grab.GrabSucceeded():
        start_time = time.time()  # and grab?
        img = grab.GetArray()
        #ms = time.time() * 1000
        ts = grab.GetTimeStamp() / 1e6  # nanoseconds to milliseconds
        frame_time_lapsed = (time.time()-start_time)*1000
        print(f'{frame_time_lapsed=}')
        #print(f'Grabbed frame {frameNum}')
        queue.put( (frameNum, ts, img) )
            
        grab.Release()

        frameNum += 1
        
    if frameNum == numOfFramesToRecord:
        break
print(f'Number of frames recorded: {frameNum}')
```

Now indeed, when I look at 'ts', the timestamp that comes from the camera, I get a 5 ms difference between any two frames.
However, when I measure the time with time.time() for recording 6000 frames, it takes about 27 seconds (instead of 30). 
When I measure the time between grabs (see code above), I get an assortment of numbers. Anywhere between 0.5 ms to 15 ms. It goes crazy. 
No I understand that my OS is not a realtime OS, but I wouldn't expect to measure 27 seconds instead of 30. It is too much.

What am I missing here?
Thanks a lot in advance,
Avi

thiesmoeller, 2022-01-19:
You are using the lowest precision time function that python provides.
If you sum up 6000 of these values you get what you see....

High precision is e.g.
https://docs.python.org/3.5/library/time.html#time.perf_counter

barliya, 2022-01-19:
Thank you for the quick reply.

I changed my timers to perf_counter.
It actually did not change anything.
I get numbers all over the place.
Also, I just ran another recording of 10 seconds (2000 frames). The recording finished within 6.3 seconds. The timestamps coming from the camera are 5 milliseconds apart and I got 2000 frames. How can this be?

I also have these in my code:
'''
camera.AcquisitionFrameRateEnable.SetValue(True)
camera.AcquisitionFrameRate.SetValue(200)
'''

also when I am checking 
'''
camera.ResultingFrameRate.GetValue()
'''
I get 200 as I should.

Maybe the grab strategy should be different?

I am really at loss here.
Thanks,
Avi

barliya, 2022-01-20:
Hi,

I think I figured out what is going on.
As I mentioned the overall run seems to be faster by few seconds, should be 30 seconds but measurements show around 27 seconds.
But your point (B) got me thinking. I don't have a memory problem because my code is multithreaded. What actually happens is that the queue that you see there does not really hold all the images, but another thread takes care of it and make sure to take every image there whenever the queue is not empty, remove it from the queue and basically save it.

So basically, measuring a multithreaded program is a different game, I guess. I took took the code I submitted earlier and isolated it to a different clean script and it works as expected re timings.
So now I am left with how to measure times in a multithreaded code. I know it is out of the scope here, but any tips will be appreciated.
Thanks again