basler/pypylon #273: Installation issues

Beleevna, 2020-09-15:
Hi,
I would like to install pypylon for a python projet on Pycham but i got an issue. 
I use Python 3.7 and Pylon 5.1.0.
This error appear, how can i fix it ? 
[image]
Thanks

thiesmoeller, 2020-12-01:
please build according to readme... or if you are expert on travis osx builds, you could help setting up compiling the binaries for osx

jpiabrantes, 2021-09-15:
> please build according to readme... or if you are expert on travis osx builds, you could help setting up compiling the binaries for osx

How does the readme suggest we install this on osx? I couldn't find it

thiesmoeller, 2025-12-15:
This issue about passing data from event handlers to the main thread has been addressed in the comments with examples. We are closing this issue. For further questions on event handlers, please refer to the discussion or open a new issue.