basler/pypylon #274: how to send the variable value in Class ImageEventPrinter(pylon.ImageEventHandler) to the main thread?

ningjingzhiwei, 2020-09-17:
Hello, everyone,
I want to process the image data in Class ImageEventPrinter, and send the data to the main thread, referring to the demo"grabusinggrabloopthread.py", but it did not work. Can anyone help? Thanks in advance.

ningjingzhiwei, 2020-09-17:
@stefanklug

thiesmoeller, 2020-10-03:
what is not working?
You have to use a datastructure that is capable to handle multi-threaded access like a queue.
Then you simply push the data ,that you generate in "OnImageGrabbed" into the queue.
And in the main-thread you wait/query on image data available and than pop this data out of the queue

thiesmoeller, 2020-12-01:
@ningjingzhiwei  ... any feedback ?

thiesmoeller, 2025-12-15:
This installation issue on macOS requires more information about the specific error and environment. Since we haven't received updates, we are closing this issue. If you are still experiencing installation problems, please open a new issue with detailed error messages and system information.