basler/pypylon #275: Wheel vesrion not working

krishna6345, 2020-09-17:
I have installed pylon 6.1.1 version on Linux 20.04.1 . Having python 3.8 version.  I am trying to install latest version of wheel but showing error as platform is not suitable for this.

scipioni, 2020-09-18:
try to usa a different wheel name pypylon-1.6.0-py3-none-any.whl

`wget https://github.com/basler/pypylon/releases/download/1.6.0/pypylon-1.6.0-cp38-cp38-linux_aarch64.whl -O /tmp/pypylon-1.6.0-py3-none-any.whl

pip install /tmp/pypylon-1.6.0-py3-none-any.whl`

thiesmoeller, 2021-02-23:
Beginning with release 1.7.1 we have pip installable versions on all platforms now.

update your pip version before by calling `python -m pip install --upgrade pip`

closed 2021-02-23 by thiesmoeller