basler/pypylon #276: clicked image containes has line patches. This happens in ~~ 5 / 50 images.

deepampatel, 2020-09-18:
[image]

thiesmoeller, 2020-09-18:
Please provide more information about your environment. 
OS. Camera model. Results of running the exact same use-case in exact same environment under pylon viewer.

EngineerFSLT, 2020-09-30:
I would guess that it's MTU size related problem, was having that when it was on default MTU = 1500.
https://www.baslerweb.com/en/sales-support/knowledge-base/frequently-asked-questions/can-you-tell-me-how-to-optimize-network-parameters-when-i-m-using-the-basler-gige-vision-filter-driver-with-my-basler-gige-camera/14985/ this should help a bit in any other case, not entirely sure what might cause this.

closed 2020-11-06 by deepampatel