basler/pypylon #279: Reading values from Camera

pmdbpc, 2020-10-01:
I am unable to read the values in readable format in python. On running the sample/call.py code with some modifications, I get the output but I'm not sure as to how to convert the data in readable format.

Sample Code: 

> from pypylon import pylon
> 
> cam = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
> cam.Open()
> 
> print(cam.GainRaw)
> cam.GainRaw.Value = 3
> print(cam.GainRaw)
>
> cam.Close()
> print("done")

To this I'm getting an output as: 
`
<pypylon.genicam.IInteger; proxy of <Swig Object of type 'GENAPI_NAMESPACE::IInteger *' at 0x0000024CC5389450> >
<pypylon.genicam.IInteger; proxy of <Swig Object of type 'GENAPI_NAMESPACE::IInteger *' at 0x0000024CC5389450> >
done
`

I understand the data ("3") is written but I cannot know for sure what's being written and what was the original data before write operation. I'm not sure if I'm reading the swig object correctly.

I'm using pycharm IDE and have the swig path setup in environment variables.

thiesmoeller, 2020-10-01:
`print(cam.GainRaw.Value)`

TiernoGs, 2020-10-10:
Use `cam.GainRaw.GetValue()` and `cam.GainRaw.SetValue(3)` instead.

pmdbpc, 2020-10-22:
Thanks for helping @thiesmoeller @TiernoGs.

closed 2020-10-22 by pmdbpc