basler/pypylon #281: Traceback error: Failed to read memory at 0x40668, 0x4 bytes. The control channel is already disconnected. (0xE1000016) : RuntimeException thrown (file 'GxPort.cpp', line 91)

AbhishekDarana, 2020-10-04:
So  I have been running cameras multiple times and then I encounter this error
Traceback error: Failed to read memory at 0x40668, 0x4 bytes. The control channel is already disconnected. (0xE1000016) : RuntimeException thrown (file 'GxPort.cpp', line 91)
Could you just help me out with what this is actually?

thiesmoeller, 2020-12-01:
do you encounter the same in pylonviewer?
typically this is due to network settings errors

AbhishekDarana, 2021-02-03:
When using pylonviewer i use camera in continuous mode....even then i dont see such issues

martonmiklos, 2021-07-15:
Hi folks, 
I am facing with the same issue by using the pylon C++ SDK and really have no idea why it get thrown when I call CInstantCamera::StartGrabbing Under what circumstances does it happen for you?

closed 2025-12-15 by thiesmoeller