basler/pypylon #282: trigger mode turns "off" when I start grabbing

niallodowd, 2020-10-05:
Hey Community,

I'm using hardware triggering. In python, I set the trigger mode of my camera to "on", but then when I start grabbing, it turns back to "off". Is there a special command to grab images using the hardware trigger? The triggering works fine in pylon viewer.

camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())

converter = pylon.ImageFormatConverter()

# Grabing Continusely (video) with minimal delay
im_num=6
k = cv2.waitKey(1)
frame=np.zeros([im_num,1642,2302])

camera.Open()
camera.TriggerMode.SetValue("On")
camera.Close()
camera.StartGrabbing(pylon.GrabStrategy_LatestImageOnly) 

idx=0
while camera.IsGrabbing():
    grabResult = camera.RetrieveResult(2000, pylon.TimeoutHandling_ThrowException)
    if grabResult.GrabSucceeded():
        idx+=1

        image = converter.Convert(grabResult)
        frame[idx] = image.GetArray()

    if idx==(im_num-1):
        break
grabResult.Release()
# When everything done, release the capture
camera.StopGrabbing()

thiesmoeller, 2020-10-05:
Only close the camera handle at the end of your program!

The way you wrote it, pylon has to reopen the device on start grabbing ... And in this step resets your Triggermode .....

niallodowd, 2020-10-05:
Yes Yes Yes!! Thank you, this solved my issue, great! I will change all of my code to only close the camera once I'm finished!!

closed 2020-10-06 by thiesmoeller