basler/pypylon #283: Connecting external light source to Basler ace camera

Terizian, 2020-10-12:
I am trying to programmatically trigger a third party light source using my camera's GPIO connections according to this page: https://docs.baslerweb.com/user-output-value

My code compiles and doesn't throw any errors, but the light source doesn't start up. I was wondering if there is any documentation of the proper set up of a light source with a Basler camera. I have the Basler ace acA4112-30um model.

thiesmoeller, 2020-12-01:
have you checked enabling/disabling the gpio in pylonviewer.

any output can be set to be used by "useroutputvalue"
https://docs.baslerweb.com/user-output-value

thiesmoeller, 2020-12-02:
any feedback @Terizian ?

Terizian, 2020-12-02:
> any feedback @Terizian ?

Hi there, thanks for your suggestion. I won't be able to try until next week. I'll let you know as soon as I can

sravan-greyscaleai, 2022-01-31:
> > any feedback @Terizian ?
> 
> Hi there, thanks for your suggestion. I won't be able to try until next week. I'll let you know as soon as I can

@Terizian Any update on this? I'm attempting to do something similar and I'm in the same situation.

thiesmoeller, 2025-12-15:
This issue about connecting external light sources has been addressed in the comments with guidance on GPIO usage. We are closing this issue. For further questions on GPIO and external devices, please refer to the Pylon SDK documentation or open a new issue.