basler/pypylon #287: Getting WaitTriggerReady time out while using grabloop method

sgn87, 2020-10-17:
I am only getting 10 frames when using camera wait for trigger function with grab loop method. 
Sample code and its output attached:
[image]

The number of frames increases if I use time.sleep(0.007) between the software triggers.
Could someone help me understand what am I doing wrong?
below is the code which I am running:

import cv2
import time
import os
from pypylon import pylon
#for recording time limit
import queue
import threading
import numpy as np
converter=pylon.ImageFormatConverter()
converter.OutputPixelFormat = pylon.PixelType_BGR8packed
converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned
img_q=None
img_number=-1
prev_img_number=img_number
#os.environ["PYLON_CAMEMU"] = "1"
class SGNImageEventHandler(pylon.ImageEventHandler):
    def OnImageGrabbed(self, camera, grabResult):
        global img_q,converter,img_number
        if grabResult.GrabSucceeded():
            if int(grabResult.ImageNumber)!=img_number:
                img_q=converter.Convert(grabResult).GetArray()
                #cv2.imshow("Frame",img_q)
                #cv2.waitKey(1)
                img_number=int(grabResult.ImageNumber)
                print("Image number:",img_number)

camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
camera.Open()
camera.RegisterConfiguration(pylon.SoftwareTriggerConfiguration(), pylon.RegistrationMode_ReplaceAll,pylon.Cleanup_Delete)    
camera.RegisterImageEventHandler(SGNImageEventHandler(),pylon.RegistrationMode_Append, pylon.Cleanup_Delete)        
camera.StartGrabbing(pylon.GrabStrategy_LatestImageOnly, pylon.GrabLoop_ProvidedByInstantCamera)
camera.TriggerMode="On"
start_rec=int(time.time())
DUR=60
while int(time.time())-start_rec<=DUR:
    try:
        if camera.IsGrabbing() and camera.WaitForFrameTriggerReady(5000,pylon.TimeoutHandling_ThrowException):
            camera.ExecuteSoftwareTrigger()
            #time.sleep(0.01)
    except Exception as e:
        print(e)
print("End")
camera.Close()

sgn87, 2020-10-17:
Also if I remove "camera.TriggerMode="On"" and run the code generates timeout error.

thiesmoeller, 2020-10-17:
below fixed code and some hints.
* configure the trigger configuration manually. Then you can work along the documentation from docs.baslerweb.com
* you got some frame "by accident", because camera started free-running and in the time between StartGrabbing and TriggerMode=On the pylon grab engine acquired ~10 frames ... so you have a fast camera ;-)
* possibly you found an issue in the SoftwareTriggerConfiguration handler....

```python
import time
from pypylon import pylon
# for recording time limit
import queue
import threading
import numpy as np
import os

converter = pylon.ImageFormatConverter()
converter.OutputPixelFormat = pylon.PixelType_BGR8packed
converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned
img_q = None

#os.environ["PYLON_CAMEMU"] = "1"

class SGNImageEventHandler(pylon.ImageEventHandler):
    def OnImageGrabbed(self, cam, grabResult):
        global img_q, converter
        if grabResult.GrabSucceeded():
            img_q = converter.Convert(grabResult).GetArray()
            # cv2.imshow("Frame",img_q)
            # cv2.waitKey(1)
        # every time you get called, you have a fresh image ...
        # no need to cross check
        print("Image number:", grabResult.ImageNumber)
        return True

camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
camera.Open()
camera.RegisterImageEventHandler(SGNImageEventHandler(), pylon.RegistrationMode_Append, pylon.Cleanup_Delete)
# configure trigger configuration manually to have full control!
camera.TriggerSelector = "FrameStart"
camera.TriggerSource = "Software"
camera.AcquisitionStatusSelector = "FrameTriggerWait"
# enable trigger mode before StartGrabbing! ... otherwise the camera will start free-running
camera.TriggerMode = "On"
camera.StartGrabbing(pylon.GrabStrategy_LatestImageOnly, pylon.GrabLoop_ProvidedByInstantCamera)

start_rec = int(time.time())
DUR = 60
while int(time.time()) - start_rec <= DUR:
    try:
        if camera.IsGrabbing() and camera.WaitForFrameTriggerReady(5000, pylon.TimeoutHandling_ThrowException):
            print("trigger")
            camera.ExecuteSoftwareTrigger()
            print(camera.AcquisitionStatus.Value)
    except Exception as e:
        print(e)

print("End")

camera.Close()

```

sgn87, 2020-10-18:
> below fixed code and some hints.
>
> * configure the trigger configuration manually. Then you can work along the documentation from docs.baslerweb.com
> * you got some frame "by accident", because camera [...]

Thanks a lot. This has removed the errors and the grabbing is happening but I am getting only 20-25 fps. I have a 200 fps camera and need atleast 120 fps. Would be really helpful if you could guide me how to achieve this. I note that the conversion process takes a lot of time but i need colored images. Saving grabResult.Array is faster but not able to get color images. Could you guide me how to go about it?

sgn87, 2020-10-18:
Also tried the following code :

#MV record module
import cv2
import time
import os
from pypylon import pylon
#for recording time limit
import queue
from datetime import datetime, timedelta, date
import numpy as np
converter=pylon.ImageFormatConverter()
converter.OutputPixelFormat = pylon.PixelType_BGR8packed
converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned
img_q=None
img_number=-1
prev_img_number=img_number
#os.environ["PYLON_CAMEMU"] = "1"
async_q=queue.Queue()

class SGNImageEventHandler(pylon.ImageEventHandler):
    def OnImageGrabbed(self, camera, grabResult):
        global img_q,converter,img_number
        if grabResult.GrabSucceeded() and int(grabResult.ImageNumber)!=img_number:
            async_q.put(converter.Convert(grabResult).GetArray())
            img_number=int(grabResult.ImageNumber)
            #print(img_number)
            
serial_number = '23431190'
info = None
for i in pylon.TlFactory.GetInstance().EnumerateDevices():
    if i.GetSerialNumber() == serial_number:
        info = i
        break
camera=pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice(info))
camera.Open()
#camera.TriggerMode="On"
#camera.TriggerSource="Software"
#camera.AcquisitionStatusSelector="FrameTriggerWait"
#camera.TriggerSelector="FrameStart"
#camera.ExposureTime=5000
camera.SensorReadoutMode="Fast"
camera.PixelFormat="BGR8"
cam_emulation=False

if cam_emulation:
    camera.ImageFilename="/home/sgn/Insightzz_code/fps_chk/pylon_emu_frames"
    camera.ImageFileMode = "On"
    camera.TestImageSelector = "Off"
    camera.TriggerSelector="AcquisitionStart"
    #camera.ExposureTimeRaw=2000
    #camera.AcquisitionFrameRateAbs=200
    
camera.RegisterConfiguration(pylon.SoftwareTriggerConfiguration(), pylon.RegistrationMode_ReplaceAll,pylon.Cleanup_Delete)    
camera.RegisterImageEventHandler(SGNImageEventHandler(),pylon.RegistrationMode_Append, pylon.Cleanup_Delete)        
camera.StartGrabbing(pylon.GrabStrategy_LatestImageOnly, pylon.GrabLoop_ProvidedByInstantCamera)
start_rec=int(time.time())
DUR=60
while int(time.time())-start_rec<=DUR:
    try:
        if camera.IsGrabbing():# and camera.WaitForFrameTriggerReady(200,pylon.TimeoutHandling_ThrowException):
            #camera.ExecuteSoftwareTrigger()
            time.sleep(0.007)
    except Exception as e:
        print(e)
print("End")
camera.Close()
print("Q Size:",async_q.qsize())
fps=int(async_q.qsize()/DUR)
img=async_q.get()
frame_width=img.shape[1]
frame_height=img.shape[0]
fourcc=cv2.VideoWriter_fourcc('H','2','6','4')
saved_vid_file="TEST_NEW_"+str(int(time.time()))+".avi"
video_writer=cv2.VideoWriter(saved_vid_file,fourcc,fps,(frame_width,frame_height))    
video_writer.write(img)
while not async_q.empty():
    img=async_q.get()
    video_writer.write(img)
video_writer.release()

With above code I am getting approx 55-60 fps but dont know if its safe for the camera and if its stable. The computer hangged for couple of second with this

thiesmoeller, 2020-10-18:
A) exact camera model ?
    You have to choose a video format that reduces the conversion complexity
B) why do you use Software Trigger at all. 
    You can configure the camera to output exactly 120fps .
    If it has to be synchronized ... You have to use hardware trigger

sgn87, 2020-10-18:
> A) exact camera model ?
> You have to choose a video format that reduces the conversion complexity
> B) why do you use Software Trigger at all.
> You can configure the camera to output exactly 120fps .
> If it has to be synchronized ... You have to use hardware trigger

A) acA1300-200uc
B) I am new to this and was referring to the grab loop sample. Now I understand that the ImageEvenhandler is called once grabbing starts in background and there is no need to software triggers. Will check out references for hardware triggers.

I am also wondering if there is a faster way to store grabresult data in a list/queue etc? As its a smart reference pointer just saving grabResult directly is of no use

sgn87, 2020-10-18:
Got the code working at full camera fps  i.e 200 fps :)
Below is the code to get that fps

[machine_vision_fullfps_recording.txt](https://github.com/basler/pypylon/files/5397605/machine_vision_fullfps_recording.txt)

closed 2020-10-18 by sgn87

thiesmoeller, 2020-10-18:
To best balance between frame rate and cpu load on your host system, is YUV pixel format.

sgn87, 2020-10-19:
> To best balance between frame rate and cpu load on your host system, is YUV pixel format.

But my camera doesnot support YUV format. The load is due to the queuing architecture so it goes little unstable when recording 3 minute video at 200 fps.

thiesmoeller, 2020-10-19:
On your device named YCbCr422_8

sgn87, 2020-10-19:
> On your device named YCbCr422_8

Got it thanks a lot.:)

MhdKAT, 2022-06-02:
> below fixed code and some hints.
>
> * configure the trigger configuration manually. Then you can work along the documentation from docs.baslerweb.com
> * you got some frame "by accident", because camera [...]

Hi I tried to use this code with a balser racer camera raL4096 but haven't been able to debug it. When I use it as it is I keep on getting "False" from cam.AcquisitionStatus.Value and when I tried to adapt to it to trigger with LineStart instead of FrameStart I kept on having "Waiting for frame trigger ready timed out " . Am I missing any parameter configuration ? and if I want to set multiple SW trigger let's say FrameStart and LineStart how do I Execute those triggers having only one function to do that camera.Execute.SoftwareTrigger(). Thnx!