basler/pypylon #288: Camera not running in Live mode using Tkinter canvas window.

pmdbpc, 2020-10-22:
I'm built a simple `tkinter` window and working on running camera in single image capture and in live mode. I have been able to capture image snapshot but the live mode doesn't seem to be working. The `tkinter` window freezes and I had to kill the window every-time. 

Is the grab strategy used for live mode correct or should I used a different grab strategy?
I have not changed the max buffer value. I have left it at default setting?
I've tried out with 10000 and 5000 as timeout value during `RetrieveResult` but nothing changes in output.

I have tried using 2 cameras - acA1600-60gm and acA1920-40gc, both giving similar results. I have attached my code files for reference. 

Both are python files. Extension changed for uploading
[camTest.txt](https://github.com/basler/pypylon/files/5421921/camTest.txt)
[camTestWindow.txt](https://github.com/basler/pypylon/files/5421922/camTestWindow.txt)

thiesmoeller, 2020-10-22:
Methods called from the gui alsways have to be short ... otherwise you lockup gui thread.

See attached small patch that gives you a live view 
.... just quick edit  as I don't really know tkinter ;-)

This is the pattern, that you should follow.

[live_view_sample.patch.txt](https://github.com/basler/pypylon/files/5425572/live_view_sample.patch.txt)

pmdbpc, 2020-10-23:
@thiesmoeller thanks for the patch file. this really helped. The live mode started but it seems there are lot of frames getting dropped. They follow a pattern the no. of frames missed slowly reduce to none and then after a couple of seconds this pattern repeats

[image]

Is it related to the frame rate? or something else?

thiesmoeller, 2020-10-23:
Maybe tkinter is queuing the updates to the image.
Try reducing camera framerate

thiesmoeller, 2020-10-23:
https://solarianprogrammer.com/2018/04/21/python-opencv-show-video-tkinter-window/

pmdbpc, 2020-10-26:
@thiesmoeller That worked I adjusted the frame rate to 10 and 98% video seems to appear good. although once in a while I do get a frame miss and see a white screen flash. I've added few more lines of code in start and stop live methods. Do you think removing these lines to another method should help here?

Also, while testing the start / stop live mode switching back to back -

sometimes (after starting / stopping live for more than 3-4 times) for start live the window gets crashed while executing:
`self.display_update_handler_hdl = self.vid.camera.RegisterImageEventHandler(self.display_update_handler, pylon.RegistrationMode_ReplaceAll, pylon.Cleanup_None)`

and for stop live, the window gets crashed while executing:
`self.vid.camera.DeregisterImageEventHandler(self.display_update_handler_hdl)`

I'm assuming the issue here is related to the handle registration / deregistration. I tried going through the code of pylon.py of pypylon, but couldn't figure out what's causing this.

thiesmoeller, 2020-11-04:
any error messages?

pmdbpc, 2020-11-05:
> any error messages?

None. 
The application window either hangs (Not Responding) or it crashes. I have put up my complete code on GitHub. 
Here is the link: [https://github.com/pmdbpc/OpenCV/tree/master/Basler%20Application](url)

The same behavior is observed when I stop camera in live mode & immediately capture an image or when I switch back to back between live mode as well. No issues when I capture back to back snapshots.

thiesmoeller, 2020-11-05:
this triggers an issue with the event handling mechanism ... can be reproduced in reduced setup.
quick workaround for you is to set ( pylon.Cleanup_Delete instead of pylon.Cleanup_None ) -> this will memleak the swig director but should keep you going for now.

pmdbpc, 2020-11-10:
@thiesmoeller - the application works much smoother now. Thanks.

closed 2020-11-10 by pmdbpc

fatmafarah, 2021-04-21:
@pmdbpc can you please post your code?

oumaymabelhajsalah96, 2021-06-29:
Hello @pmdbpc , can you help me, I work with a basler and tkinter camera like you.
I can contact you on gmail please,
my gmail is oumayma.belhajsalah96@gmail.com

Lucrezia-Cester, 2022-12-06:
Is there a code for this live script now?