basler/pypylon #289: I'm currently installed pypylon with pylon 6.1.1 version. Does pypylon support pylon v6.1.1?

Heermosi, 2020-10-26:
Help?
And during the installation, the pypylon searched the /opt/pylon5 for pylon installations. Shall it be changed later?

thiesmoeller, 2020-10-26:
pylon 6.1.1 fully supported.

This wrong search path will be corrected soon.

For the time being set PYLON_ROOT to /opt/pylon6

Heermosi, 2020-11-03:
> pylon 6.1.1 fully supported.
> 
> This wrong search path will be corrected soon.
> 
> For the time being set PYLON_ROOT to /opt/pylon6

Alright, I've made a symbol link
Besides, I've experienced some error in examples. The grabone.py cannot work well with 6.1.1. ChunkModeActive was not found in the camera settings. And should here be some tutorials about variables able to be set?

thiesmoeller, 2025-12-15:
This question about pypylon compatibility with Pylon 6.1.1 has been answered. Pypylon supports Pylon 6.x versions. We are closing this issue to keep our tracker current.