basler/pypylon #292: Zero Copy Segfault

Simon-Cochrane, 2020-10-30:
There is a consistant segfault happening in the GetArrayZeroCopy function. In particular the GetNumBufferExports produces that segfault. Since that call is only really used to verify that the user is not misusing the memory view, skipping that call unblocks my system knowing that we are not trying to misuse the memory view, but it doesn't actually solve the underlying issue.

The GetNumBufferExports constantly causes segfaults. I've been able to reproduce the segfault pretty reliability with a 5 cameras system all running concurrently. I have tested that it reproduces on pylon 5.2.0 and 6.1.1. (Latest) as well as pypylon 1.5.1 and 1.6.0 (Latest). After skipping the GetNumBufferExports call, it ran a continuous test of over 14h+ of the 5 camera system mentioned above using the latest pylon and pypylon.

I have a forked branch which implements the workaround of skipping the GetNumBufferExports based on an optional kwarg. But I wanted to raise the actual underlying issue here.

thiesmoeller, 2020-11-04:
could you please provide a sample to reproduce the segfault?

thiesmoeller, 2020-12-01:
@Simon-Cochrane can you please provide a programming sample for the segfault

Simon-Cochrane, 2020-12-01:
I'm running this as part of a bigger system, but let me give you some code snippets from the time we had this issue:

This is how each image was being grabbed.

```
img: Optional[np.array] = None
with self.camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException) as grabResult:
    if grabResult.GrabSucceeded():
        with grabResult.GetArrayZeroCopy() as zc:
            img = cv2.cvtColor(zc, color_format)
```

Each camera is running on its own thread and calling thus concurrently to all other cameras, all in the same python process. I managed to reproduce the issue on a 5 camera system with the following: 1x acA4096-30uc and 4x acA720-520uc

Some of the images are garbage collected almost as soon as they are copied, others may be kept in the system for a few seconds for extra computation (though that would happen in a different thread). In each case, the camera thread is dedicated to simply getting the image from the camera and making it available to the rest of the system. No other computation occurs on the camera thread.

Also note the color conversion, we also tried doing an np.copy there and doing the color version further down outside the zero copy context with no change to the segfaulting behavior.

As I mentioned before, we were able to solve this issue by taking the code that lives inside the GetArrayZeroCopy function and using that directly, while removing the call to GetNumBufferExports, which was the call causing segfaults.

aquilesC, 2020-12-22:
I am not sure how the ring buffer works with pylon. I was having plenty of segfaults when acquiring from two cameras (I was not using zero copy) and had to work around it implementing a Lock when using GetArray.

thiesmoeller, 2020-12-23:
@aquilesC can you reproduce the segfaults in a smaller example, that you can share.

Optimal would be, if you can reproduce, while using the emulated cameras, so we can check if this is an issue in the upper layers, or something that is connected to one of the transport layers.

calling an application with this environment variable:
`PYLON_CAMEMU=5`
will generate 5 emulated cameras.

What also would be interesting to see, if you call your python interpreter like so:

`PYTHONMALLOC=pymalloc_debug python3 your_crash_sample.py`

or

`PYTHONMALLOC=malloc_debug python3 your_crash_sample.py`

This will enable all debug hooks in the python memory manager.

thiesmoeller, 2020-12-23:
@Simon-Cochrane do you manage each camera completely in their respective thread?
So Creation/Opening/Parametrization/Grabbing all happens in their respective thread, and only the import of pypylon ( with the underlying pylon initialization) happens from main thread ?