basler/pypylon #293: Sample code for multiple camera with hardware trigger

agosztolai, 2020-11-01:
Hello, 

I have two Basler acA1920-150um, and I have already set up the hardware trigger and was able to trigger the cameras in Pylon Viewer.

Now, I am trying to modify the `samples/grabmultiplecameras.py ` file to have hardware triggering. 

However, I am having trouble. 

First, if I run  `samples/grabmultiplecameras.py`, the code fails precisely every second time, giving me the error 

`RuntimeException: Failed to open device '2676:ba02:13:2:22' for XML file download. Error: 'Device is exclusively opened by another client.' : RuntimeException thrown (file 'PylonUsbTL.cpp', line 474)`

Could help with this? 

Also, could you please provide a hint/sample code for how to modify `samples/grabmultiplecameras.py` to incorporate hardware trigger?

agosztolai, 2020-11-01:
Below is what I managed so far. 

```
from pypylon import pylon
from pypylon import genicam
import sys

countOfImagesToGrab = 2 # Number of images to be grabbed.
maxCamerasToUse = 2 # Limits the amount of cameras used for grabbing.
exitCode = 0 # The exit code of the sample application.

try:
    
    tlFactory = pylon.TlFactory.GetInstance()
    
    # Get all attached devices and exit application if no device is found.
    devices = tlFactory.EnumerateDevices()
    if len(devices) == 0:
        raise pylon.RuntimeException("No camera present.")
        
    # Create an array of instant cameras for the found devices and avoid exceeding a maximum number of devices.
    cameras = pylon.InstantCameraArray(min(len(devices), maxCamerasToUse))
        
    for i, camera in enumerate(cameras):
        
        # Create an instant camera object
        camera.Attach(tlFactory.CreateDevice(devices[0]))
    
        #camera name
        camera_name = '_'.join([camera.GetDeviceInfo().GetModelName(),
                                camera.GetDeviceInfo().GetSerialNumber()])
        
        # pylon.FeaturePersistence.Load('./camera_params/' + cam_name + '.pfs', camera.GetNodeMap())
        
        camera.Open()

        # Print the model name of the camera.
        print("Using device ", camera_name)

        camera.MaxNumBuffer = 5 #count of buffers allocated for grabbing
        camera.TriggerSelector.SetValue('FrameStart')
        camera.TriggerMode.SetValue('On') #hardware trigger
        camera.TriggerSource.SetValue('Line1')
        camera.TriggerActivation.SetValue('RisingEdge')
        camera.StartGrabbingMax(countOfImagesToGrab)
        camera.ExposureAuto.SetValue('Off')
        camera.ExposureTime.SetValue(200)
  
        while camera.IsGrabbing():
            # Wait for an image and then retrieve it. A timeout of 5000 ms is used.
            grabResult = camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)

            print("GrabSucceeded: ", grabResult.GrabSucceeded())
            grabResult.Release()
        camera.Close()

except genicam.GenericException as e:
    # Error handling.
    print("An exception occurred.")
    print(e.GetDescription())
    exitCode = 1

sys.exit(exitCode)
```

This returns output but not the correct one. 

```
Using device  acA1920-150um_40018632
GrabSucceeded:  True
GrabSucceeded:  True
Using device  acA1920-150um_40018632
GrabSucceeded:  True
```

The problem is that the cameras are opened one after the other and hence the second camera misses the first trigger. Hence only one frame is grabbed.

**How do you ensure that all cameras are open at the same time and will receive the triggers?** Then the readout could be done in a loop.

Also, I'd like to read in the camera params from a config file (see the commented line). However, this does not work. 

**Could you please show how to do this correctly?**

agosztolai, 2020-11-02:
I wanted to post a sample code that worked for me so far. 

Importantly, I found that on macOS (at least), `EnumerateDevices()` returns a number of devices even if no cameras ara attached. I got around this by checking their serial numbers against those of my cameras.

```
from pypylon import pylon
from pypylon import genicam
import numpy as np
import sys

countOfImagesToGrab = 2 # Number of images to be grabbed
maxCamerasToUse = 2 # Limits the amount of cameras used for grabbing
timeout = 5000 #max time to wait for grabbed image
serial_numbers = ['40018632', '40038871']

# The exit code of the sample application.
exitCode = 0

try:
    
    tlFactory = pylon.TlFactory.GetInstance()
    
    # get all attached devices
    devices = tlFactory.EnumerateDevices()
    
    # look for unique devices that match the serial numbers
    camera_devices = []
    for device in devices:
        serial = device.GetSerialNumber()
        
        for sn in serial_numbers:
            if serial == sn:
                camera_devices.append(device)   
                serial_numbers.remove(serial)
                break                        
              
    camera_devices = tuple(camera_devices)
    
    # exit if no camera found
    if len(camera_devices) == 0:
        raise pylon.RuntimeException("No camera present.")
        
    print(str(len(camera_devices)) + ' cameras found.')

    # Create an array of instant cameras for the found devices and avoid exceeding a maximum number of devices.
    cam_array = pylon.InstantCameraArray(min(len(camera_devices), maxCamerasToUse))
        
    for i, camera in enumerate(cam_array):
        
        # Create an instant camera object
        camera.Attach(tlFactory.CreateDevice(camera_devices[i]))
    
        #camera name
        camera_name = '_'.join([camera.GetDeviceInfo().GetModelName(),
                                camera.GetDeviceInfo().GetSerialNumber()])
      
        camera.Open()

        # set camera parameters
        camera.MaxNumBuffer = 5 #count of buffers allocated for grabbing
        camera.TriggerSelector.SetValue('FrameStart')
        camera.TriggerMode.SetValue('On') #hardware trigger
        camera.TriggerSource.SetValue('Line1')
        camera.TriggerActivation.SetValue('RisingEdge')
        camera.StartGrabbingMax(countOfImagesToGrab)
        camera.ExposureAuto.SetValue('Off')
        camera.ExposureTime.SetValue(200)
        
        # Print the model name of the camera.
        print("Setup of camera ", camera_name, " complete.")
        
except genicam.GenericException as e:
    # Error handling.
    print("An exception occurred.")
    print(e.GetDescription())
    exitCode = 1        
    
try:  
    for i, camera in enumerate(cam_array):
        while camera.IsGrabbing():
            
            # Wait for an image and then retrieve it. A timeout of 5000 ms is used.
            grabResult = camera.RetrieveResult(timeout, pylon.TimeoutHandling_ThrowException)

            print("GrabSucceeded: ", grabResult.GrabSucceeded())

            grabResult.Release()
        camera.Close()

except genicam.GenericException as e:
    # Error handling.
    print("An exception occurred.")
    print(e.GetDescription())
    exitCode = 1

sys.exit(exitCode)
```

closed 2020-11-03 by agosztolai

farhad-dalirani, 2022-07-26:
How did you set two Basler cameras as master-slave? Where in pylon viewer I can set master camera generate trigger for the slave camera?