basler/pypylon #295: Is parallelising the image grabbing and saving from multiple cameras worth it?

agosztolai, 2020-11-03:
Hello, 

I would greatly appreciate any advice on the following.

I have implemented your code with a hardware trigger for multiple cameras.

I would like to push the framerate high for potentially extended periods. I am currently appending the grabbed frames into an array from all cameras before saving, which is quite CPU intensive. So I considered parallelizing.

Do you have experience with this and do you know if it could result in a speedup?
What is the fastest way to save images from multiple cameras?

I have tried using multiprocessing and passing each camera to a different worker

```
def read_cam(camera):
    
    imgs_cam = []
    count = 0
    try:
        while camera.IsGrabbing():
            grabResult = camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)

            imgs_cam.append(grabResult)

            grabResult.Release()
            
    except genicam.GenericException as e:
        print("Some frames have been dropped.")
        print(e.GetDescription())
        
    return imgs_cam

def grab_frames(cam_array):

    jobs = []
    for i, camera in enumerate(cam_array):
         process = multiprocessing.Process(target=read_cam, 
                                           args=(camera,))
         jobs.append(process)
        
    # Start the processes     
    for j in jobs:
         j.start()

    # Ensure all of the processes have finished
    for j in jobs:
        j.join()
```

However, this is not possible because the camera is SwigPyObject, which cannot be pickled and sent to the workers.

robinvdneut, 2020-11-27:
No, generally not in Python. Parallelising is useful, when you need to wait for something, while other things should run continuously, but in most cases I use signals for that. But you will only increase speed, if you can use multiple processor cores, which python barely allows. Normal threading just uses one core, while one single processor switches between the threads.

This means you can better improve your appending algorithm. _list.append()_ is not really efficient, as it computers like to reserve a defined amount of space for each object in advance, so in the background, the _list_-object needs to be redifined completely when appending.

see this: [https://stackoverflow.com/questions/7542957/is-python-capable-of-running-on-multiple-cores](https://stackoverflow.com/questions/7542957/is-python-capable-of-running-on-multiple-cores)

If you don't mind defining a maximum number of images to save, you could try to use a _numpy.ndarray_ with 3 dimensions, respectively 4 dimensions with colours.

My impression is, that by far the ugliest part of my loop is plotting data, e.g. the image, even using Qt's C-Compiled libs. Followed by the grabbing itself.

thiesmoeller, 2020-11-30:
running the image processing on multiple cores makes absolute sense.
you should split your program into one thread that handles all image acquisition and 
then use multiprocessing to exchange the acquired frames with workers on other cores.
with this setup, you don't have to sync the complex InstantCamera object between processes.
Only ndarrays ( images ) which are simple.

aquilesC, 2020-12-22:
I currently use PyZMQ to share images from the acquisition process to the worker processes (including savers), you can see the saver [here](https://github.com/aquilesC/disperscripts/blob/master/calibration/models/movie_saver.py), and the camera streaming data [here](https://github.com/aquilesC/experimentor/blob/b41c4b07f01be233f05c681b0d5677e6626c166d/experimentor/models/devices/cameras/basler/basler.py#L372). It is a great way of leveraging multi-core computers, but you must be sure workers can keep up, it is tricky to handle overrunning with ZMQ.

thiesmoeller, 2025-12-15:
This question about parallelizing image grabbing and saving has been addressed in the comments with guidance on threading vs multiprocessing. We are closing this issue. For further performance optimization questions, please refer to the discussion or open a new issue.

closed 2025-12-15 by thiesmoeller