basler/pypylon #298: Intermittent "ValueError : Pixel format currently not supported"

bwllc, 2020-11-13:
I have constructed a PyPylon application which works with up to three acA2500-14um cameras at once in continuous acquisition mode.  One of my three cameras appears to be having issues, and I am not sure whether the problem is software or hardware.

Frames from each camera are grabbed by my software with the following code to PyPylon:

`return self.camera.RetrieveResult(self.HANDLE, pylon.TimeoutHandling_ThrowException).Array`

The one camera that fails can run stably for up to several minutes.  It does not appear to matter whether I am also connected the other cameras.  When the system fails, the traceback is as follows:

```
Traceback (most recent call last):
  File "C:\Users\twa2\Documents\Python Scripts\Visual navigator 0.8\views.py", line 320, in grab_array
    pylon.TimeoutHandling_ThrowException).Array
  File "C:\Users\twa2\Anaconda3\lib\site-packages\pypylon\pylon.py", line 109, in func_wrapper
    return func(*args, **kwargs)
  File "C:\Users\twa2\Anaconda3\lib\site-packages\pypylon\pylon.py", line 2113, in GetArray
    shape, dtype, format = self.GetImageFormat(pt)
  File "C:\Users\twa2\Anaconda3\lib\site-packages\pypylon\pylon.py", line 109, in func_wrapper
    return func(*args, **kwargs)
  File "C:\Users\twa2\Anaconda3\lib\site-packages\pypylon\pylon.py", line 2095, in GetImageFormat
    raise ValueError("Pixel format currently not supported")
ValueError: Pixel format currently not supported
```

According to the GetImageFormat source, I should only receive this error if my pixel format is not one of the enumerated types.  For each camera, my program calls

`camera.PixelFormat.SetValue("Mono8")`

exactly once during setup, and thereafter the pixel format is left alone.

Please help me to understand this problem.  Thanks!

thiesmoeller, 2020-11-13:
Do you check GrabSucceeded() before any access to the grab result?

bwllc, 2020-11-16:
Thank you @thiesmoeller for your reply.

I have only just discovered GrabSucceeded() because of your post.  I have been operating groups of Basler cameras connected to two different computer for several months without needing this method in my code.  Do you think I might have a hardware problem?

GrabSucceeded() is a method of a GrabResult object.  Therefore I apparently need to create a GrabResult object, which my code currently does not.  Do I need to supply arguments to the GrabResult constructor?  It appears that the "rhs" smart pointer argument list is optional, but I am not sure of its purpose, and thus I don't know whether I need to specify something.

thiesmoeller, 2020-11-19:
Hi @bwllc ,

please have a look to the programming samples ( either the one in the main readme.md file ) or the other ones in the samples/ folder, where you can see the usage of GrabSucceeded().

Great to see, that you stepped into the documentation :-)
And immediately saw the main issue with it. We autogenerate the docs from the doxygen comments in the C++ headers.
This works for _most_ of the functions quite well, although for any function, where we had to change the signature due to be pythonesque, the docs are misleading ... like you realized

Thies

bwllc, 2020-11-24:
Thank you @thiesmoeller, I have rebuilt my system to ignore the occasional bad GrabResult by including calls to GrabSucceeded().  I would still like to understand why one of our three cameras is misbehaving, but our system can still function with the occasional lost video frame.

closed 2020-11-24 by bwllc