basler/pypylon #299: How do I configure (set values) on multiple cameras in a python script

witekgb, 2020-11-13:
I am having problems trying to configure two cameras attached to my computer.  As a template I am using the grabmultiplecameras.py sample script. 

If I query the cameras for their serial number in this way:

``serial_number_one = cameras[0].GetDeviceInfo().GetPropertyValue('SerialNumber')``
``serial_number_two = `cameras[1].GetDeviceInfo().GetPropertyValue('SerialNumber')``

After this for loop:
``    for i, cam in enumerate(cameras):``
``       cam.Attach(tlFactory.CreateDevice(devices[i]))``

I get the correct answer.  However if I try to set the exposure time in the same location, using the same format (treating the "cameras" as an array):

``cameras[0].ExposureTime.SetValue(2000)``

I get this error:

Traceback (most recent call last):
  File "C:/Users/User/Documents/BatterUp/Python/Grab_MultipleCameras.py", line 63, in <module>
    cameras[0].ExposureTime.SetValue(2000)
  File "C:\Users\User\AppData\Local\Programs\Python\Python37-32\lib\site-packages\pypylon\genicam.py", line 2217, in SetValue
    return _genicam.IFloat_SetValue(self, Value, Verify)
_genicam.AccessException: Node is not writable. : AccessException thrown in node 'ExposureTime' while calling 'ExposureTime.SetValue()' (file 'floatt.h', line 85)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:/Users/User/Documents/BatterUp/Python/Grab_MultipleCameras.py", line 100, in <module>
    print("An exception occurred.", e.GetDescription())
AttributeError: 'AccessException' object has no attribute 'GetDescription'

Any help is appreciated.

thiesmoeller, 2020-11-14:
Call cameras[0].Open() ...

witekgb, 2020-11-14:
THANK YOU! 
``cameras[0].Open()``
``cameras[0].ExposureTime.SetValue(2000)`` 
Did it!

closed 2020-12-01 by thiesmoeller