basler/pypylon #301: Start_Processing : Exception in Start_Processing() Main Process:Failed to read memory at 0x40668, 0x4 bytes. Timeout, no message received. (0xE101800B) : RuntimeException thrown (file 'GxPort.cpp', line 91)

AbhishekDarana, 2020-11-27:
could you please help me with this issue

thiesmoeller, 2020-12-02:
please provide any form of context to your request.

from what you write only thing to see: you use a gige camera. Communication to the camera is lost.
missing: OS / platform / pypylon version / camera model / operating point ( single/multiple camera, while streaming, before streaming ... )

AbhishekDarana, 2020-12-02:
Hey I think I was able to solve the problem . I was using a multiprocess to save frames and sometimes had to wait for longer time than timeout , that was because of poor coding. Sorry for not providing information while posting  the question, wanted to post it once I was sure that i wud get a reply. Thanks!!

thiesmoeller, 2020-12-12:
With the ticket reopened:

please provide any form of context to your request.

from what you write only thing to see: you use a gige camera. Communication to the camera is lost.
missing: OS / platform / pypylon version / camera model / operating point ( single/multiple camera, while streaming, before streaming ... )

AbhishekDarana, 2021-02-04:
OS being used is ubuntu 16.04 , pypylon version is 5.1 camera  model is acA2040-35 gc. Used a numtiple camera setup twhich are triggered at 8 fps using software trigger.
System runs smoothly and then throws this error... Failed to write memory at 0x40224, 0x4 bytes. Timeout, no message received. (0xE101800B) : RuntimeException thrown (file 'GxPort.cpp', line 126).
Thank you in advance

closed 2025-12-15 by thiesmoeller