basler/pypylon #302: bcon mipi daA2500-60mci not up to true fps

ancabilloni, 2020-12-10:
Hi
I have a bcon mipi daA2500-60mci running on Jetson Nano. Per spec, this camera should have up to 60fps. But I couldn't find a way to get it more than 10fps. I tried to set exposure time to min, turn off exposure auto. But the fps still did not improve. What did I do wrong?

This is the code the I use to display with OpenCV 3.4.11

```'''
from pypylon import pylon
import time
import cv2

# conecting to the first available camera
camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
camera.Open()

#camera.Gain = camera.Gain.Max
#camera.ExposureTime.SetValue(camera.ExposureTime.Min)

camera.ExposureAuto.SetValue("Off")

# Grabing Continusely (video) with minimal delay
camera.StartGrabbing(pylon.GrabStrategy_LatestImageOnly)
converter = pylon.ImageFormatConverter()

# converting to opencv bgr format
converter.OutputPixelFormat = pylon.PixelType_BGR8packed
converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned
start = time.time()
counter = 0
while camera.IsGrabbing():
    grabResult: pylon.GrabResult = camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)

    if grabResult.GrabSucceeded():
        # Access the image data
        image = converter.Convert(grabResult)
        img = image.GetArray()
        counter += 1
        elapsed = time.time() - start
        cv2.putText(img, "{} FPS".format(int(counter / elapsed)), (200, 20),
                    cv2.FONT_HERSHEY_SIMPLEX, 0.5, (0, 0, 255), 2)
        cv2.imshow('title', img)
        k = cv2.waitKey(1)
        if k == 27:
            break
    grabResult.Release()

# Releasing the resource
camera.StopGrabbing()
camera.close()

cv2.destroyAllWindows()
```

ancabilloni, 2020-12-10:
After playing around a bit with settings, I could finally achieve higher FPS. I have to set the Width and Height lower. With the below setting, I got about 30 FPS

```
camera.Gain = camera.Gain.Max
camera.ExposureTime.SetValue(camera.ExposureTime.Min)
camera.AcquisitionFrameRate.SetValue(60)
camera.AcquisitionFrameRate.SetValue(True)
x = 200
y = max(0, int(0.75*x - 4))
w = max(96, 2592 - 8*x)
h = max(104, 1944 - 8*y)
camera.Width.SetValue(w)
camera.Height.SetValue(h)
```
FPS is even better if I go real small on the height. But I don't want to go too small. Other suggestions to improve FPS for this camera is welcome. My application is for outdoor.

FrankKarstens, 2020-12-10:
This is a bandwidth issue. The dart daA2500-60mci can only deliver the full 60 fps when connected with all four MIPI lanes (the camera can transmit max 1.2 GBit/s/per lane). Unfortunately the Nvidia Jetson Nano DevKit only provide twos MIPI lanes. That's why you get only 30 fps max with full 5MPx resolution. If you reduce the resolution (e.g. by reducing the height or width) you have to transmit less data per frame and you can transmit for frames per second.

ancabilloni, 2020-12-11:
@FrankKarstens can you elaborate more into how to get maximum 30FPS with full 5MP resolution? Because in my original problem, without reducing width and height, I can only get maximum 10 FPS with ExposureAuto off.

FrankKarstens, 2020-12-11:
It is the displaying of the image and the high CPU load in the grab loop which slows down your whole application.
You should use a named window and should reduce the display area. here's a proposal:

`from pypylon import pylon
import time
import cv2

# conecting to the first available camera
camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
camera.Open()

#camera.Gain = camera.Gain.Max
#camera.ExposureTime.SetValue(camera.ExposureTime.Min)

camera.ExposureAuto.SetValue("Off")
camera.ExposureTime.SetValue (20000)

# Grabing Continusely (video) with minimal delay
camera.StartGrabbing(pylon.GrabStrategy_LatestImageOnly)
converter = pylon.ImageFormatConverter()

# converting to opencv bgr format
converter.OutputPixelFormat = pylon.PixelType_BGR8packed
converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned
start = time.time()
counter = 0

# use OpenGL for image display if your opencv build supports OpenGL
#cv2.namedWindow('mipi test', cv2.WINDOW_NORMAL|cv2.WINDOW_OPENGL)

# otherwise just use CPU
cv2.namedWindow('mipi test', cv2.WINDOW_NORMAL)
cv2.resizeWindow('mipi test', 1200, 800)

while camera.IsGrabbing():
    grabResult: pylon.GrabResult = camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)

    if grabResult.GrabSucceeded():

        # Access the image data and convert with pypylon methods
        # image = converter.Convert(grabResult)
        # img = image.GetArray()
        
        # or alternatively convert with opencv methods
        img = cv2.cvtColor( grabResult.GetArray() , cv2.COLOR_YUV2BGR_UYVY)
        
        counter += 1
        elapsed = time.time() - start
        cv2.putText(img, "{} FPS".format(int(counter / elapsed)), (200, 20),
                    cv2.FONT_HERSHEY_SIMPLEX, 0.5, (0, 0, 255), 2)
        
        
        cv2.imshow('mipi test', img, )
        k = cv2.waitKey(1)
        if k == 27:
            break
    grabResult.Release()

# Releasing the resource
camera.StopGrabbing()
camera.close()

cv2.destroyAllWindows()`

ancabilloni, 2020-12-11:
@FrankKarstens Thank you. Switching to `img = cv2.cvtColor( grabResult.GetArray() , cv2.COLOR_YUV2BGR_UYVY)` makes a big difference. I got close to 30FPS without reducing resolution and no display, switching back to pypylon converter stays at 14 FPS for the same test. The CPU displays also bogs down the speed at 14 FPS. I don't have openGL built support, I may try that again after rebuilding OpenCV some other time.

closed 2020-12-12 by ancabilloni