basler/pypylon #303: Camera seems to freeze after changing line height.

PolymaTh-EE98, 2020-12-15:
I am using a linescan (racer) camera and need to change the frame size as product dimensions change. Using the method mentioned in this post causes issues for me.
[https://github.com/basler/pypylon/issues/145](https://github.com/basler/pypylon/issues/145)

Here is the relevant part of my code.

`camera.AcquisitionStop.Execute()
camera.TLParamsLocked = False
camera.Height = int(CameraHeight)
camera.TLParamsLocked = True
camera.AcquisitionStart.Execute()`

My work around is to set the line height to large number in pylon then saving this pfs file.  If I set the lines from Python with the code above to something smaller than what is in the pfs file the camera stops sending me images.  This is a problem as people who are not aware use Pylon Viewer and save the file over mine then the camera quits working with my Python code.

DxTa, 2024-10-25:
I seems to have similar issue with AutoFunctionROI's parameters.

I think the camera waits for the trigger signal again (https://docs.baslerweb.com/acquisition-start-stop-and-abort) and if the camera having trigger parameter = off, then it will not trigger automatically. Thus maybe changing trigger to Continuous can solve in some cases (https://docs.baslerweb.com/free-run-image-acquisition).

I will do some more tests on this.

thiesmoeller, 2025-12-15:
This issue about camera freezing after changing line height requires more information about the specific camera model and configuration. Since we haven't received updates, we are closing this issue. If you are still experiencing this problem, please open a new issue with detailed camera model and configuration information.