basler/pypylon #305: odroid xu4 with basler camera aca640-750um grap issue

Skm1221, 2020-12-16:
Hello, 

I'm developing some image processing software with odroid xu4(ubuntu 18.04 mate) and aca630-750um camera

When I grabed image with fps220, grab is failed few minutes later and reopen camera was failed. 

Then, without unplugging usb port I can't use camera. 

But, when I check lsusb message, basler camera is recognized. 

So I executed pylon viewer, but camera was not detected. 

below image is result of dmesg. 

Before camera grab fail situation happen, that message always occured. I think that is problem 

Is anybody know how to solve this problem ?

[image]

I tried to grab image with pylon viewer for removing any image processing logics.

But It was same. 
After few minutes later, camera is removed and then reopen is not possible

[image]

MCilento93, 2020-12-22:
In your code assure to close the camera before a new grabbing: on case of exception assure again camera.Close()

thiesmoeller, 2021-01-08:
... and check the quality of your USB3 cable.

On a bad USB3 cable a number low level USB errors might occur.