basler/pypylon #307: No binary for python 3.9

josselex, 2020-12-17:
It would be good to have a binary releas on pypi for python 3.9.

lars-ek, 2021-01-11:
I agree, I am also waiting...

thiesmoeller, 2021-01-11:
I assume you work with a source build at the moment?

josselex, 2021-01-12:
@thiesmoeller
No. I am holding back rollout of python 3.9 because of this,

BlackbirdB51, 2021-02-08:
I'd like to have a release as well.
I tried to build it from source, but did not manage to do it. What Visual Studio packages do I need?
I resorted to install temporarily python 3.8 as well for that specific application.

marv, 2021-02-24:
pypylon supports Python 3.9 now starting with 1.7.0 and you can find releases either here on GitHub or on PyPi 🥳

BlackbirdB51, 2021-02-24:
Yes. Initially the builds for windows were missing, but with 1.7.1 they are present now. Thanks to the devs!

closed 2021-02-24 by stefanklug