basler/pypylon #311: grabResult.GrabSucceeded() not so often

MCilento93, 2020-12-24:
### Type of Issues
Enhancement/Error

### Operating System
Windows 64

### Python  version
3.7.9

### pypylon version
1.5.4

### Basler camera
acA1300-30g

### Description of Problem / Question / Details
My target is to grab a single photo when a button is pressed in a GUI.
The **method** triggered now is the one reported below:

```python
import numpy as np
NULL_ARRAY=np.ndarray(shape=(0,0))

def grab_basler_photo():
    array=NULL_ARRAY
    try:
        from pypylon import pylon
        camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
        camera.Open()
        numberOfImagesToGrab = 100
        camera.StartGrabbingMax(numberOfImagesToGrab)
        while camera.IsGrabbing():
            grabResult = camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)
            if grabResult.GrabSucceeded():
                array = grabResult.Array
                camera.Close()
                return array
            else:
                pass
            grabResult.Release()
        camera.Close()
    except Exception as Err:
        array=NULL_ARRAY
        logger.error('errors with pypylon\n'+Err)
    return array
```

As you can see, I simply want to retreive the numpy array from camera **as soon as** the capture is successfull (no pixels above bit depth and capture without errors/common black stripes). 
I want implement the **fastest** way to do this simple one-shot grab. 

Actually it happens that often `numberOfImagesToGrab = 100` is **not sufficient** i.e. `grabResult.GrabSucceeded()` is most of the times false, I need to increase this buffer a lot and it takes too much time for my goals.

Thank you for everybody willing to help, and have nice Xmas holidays.

thiesmoeller, 2020-12-31:
Hi @MCilento93 , grabbing _must_ succeed  every time with correct settings!

What you describe ( black stripes ) results from wrong bandwidth configuration.
 Please read the chapter on network parameters: https://docs.baslerweb.com/network-related-parameters-(gige-cameras)
You can verify correct settings in the pylon viewer with the bandwidth manager: https://docs.baslerweb.com/bandwidth-manager#optimizing-bandwidth-consumption

Bandwidth manager will show you most issues, that might exists in your setup.

MCilento93, 2021-01-01:
> Hi @MCilento93 , grabbing _must_ succeed  every time with correct settings!
>
> What you describe ( black stripes ) results from wrong bandwidth configuration.
>  Please read the chapter on network [...]

Thank you for the answer,
I will give a look as soon as I can.

On **pylon viewer** I did not checked the bandwitch tool, even though in there for single shots and continuous capture didn't give any errors.

nchauhan5, 2021-03-16:
@thiesmoeller - I have the correct settings as checked from the bandwidth manager. I have the Basler aca4600-7gc Ace GigE Camera. I have the same problem. The statement - if grabResult.GrabSucceeded() is false more often than not. Like 95% of the time. This Camera at my current settings is giving out 2 frames per second as seen in the pylon viewer, but I am getting like 1 frame every 10 seconds or something.

MCilento93, 2021-03-16:
> @thiesmoeller - I have the correct settings as checked from the bandwidth manager. I have the Basler aca4600-7gc Ace GigE Camera. I have the same problem. The statement - if [...]

Are you trying my code above?
I had at the time many troubles with ethernet ports and cables... With right settings if worked

nchauhan5, 2021-03-16:
@MCilento93 - I got this to work just now. Make sure you add the statement "camera.Open()" before calling "camera.StartGrabbing". I was able to get more frames then before at the frame rate that my camera supports at the settings that I have. Try it now. Seems you are still actively looking for an answer on this @MCilento93

thiesmoeller, 2025-12-15:
This issue about rarely getting successful grabs requires more information about camera settings, exposure time, and trigger configuration. Since we haven't received updates, we are closing this issue. If you are still experiencing this problem, please open a new issue with detailed camera configuration.