basler/pypylon #312: Empty import

jeremytieman, 2020-12-29:
I have Pylon 5.2 and Python 3.7 installed on Windows 10. When I run the Pylon Viewer, I can see and capture images from the Basler USB camera I have attached to the computer. I downloaded the pypylon-1.4.0-cp37-cp37m-win_amd64.whl file and installed it using pip. When I run Python and import pypylon, all I get is an empty import:

`['__builtins__', '__cached__', '__doc__', '__file__', '__loader__', '__name__', '__package__', '__path__', 
'__spec__']`

Running python -v doesn't give me any errors or any way to further troubleshoot. Can anyone give me any ideas or extra suggestions to find out what is going wrong?

thiesmoeller, 2020-12-29:
Check the grab sample in the readme on the main page.

... and you should use the latest version ...
https://github.com/basler/pypylon/releases/download/1.6.0/pypylon-1.6.0-cp37-cp37m-win_amd64.whl

jeremytieman, 2020-12-29:
At the moment I can't use the latest version, I am version constrained to the developed software. That being said, I think it is now working because I was importing pypylon as pylon rather than importing pylon from pypylon. Closing issue.

closed 2020-12-29 by jeremytieman