basler/pypylon #313: Memory Accumulation

zamarjavier, 2021-01-04:
Hello,

I'm have two Usb camera basler and I'm trying to obtain a sinchronization capture using a software trigger. After that, I process the images in each camera. The idea is to execute multiple launches doing this work for each launch. I achieve this goal but now, I have another problem. I have a memory accumulation. 
For example after running the code my RAM accumulates 3GB. Then after the first launch, the RAM accumulates 5,6. After the second launch the RAM accumulates 8,2.

 I'm using threads in python. 2 for each camera, one for capturing and the other for processing.
I don't know how to solve this problem, does anybody know how to do it?

Thanks in advance.

Kind regards

zamarjavier, 2021-01-05:
Thanks for your response @thiesmoeller

I'm using Ubuntu 18 LTS. 

I think that I have detected the problem but  I don't know how to solve it.
I'm capturing 600 images for each camera and the memory grows when I extract the image form grabresult and insert into  a list. Here you can see an example:

a = []
while camera.IsGrabbing():
      grab_result = camera.RetrieveResult(10000, pylon.TimeoutHandling_ThrowException)
      if grab_result.GrabSucceeded():
             a.append(grab_result.GetArray()

I need the images into a list, because when de capture process ends, I need to process them.

Do you know how to do it whithout memory accumulation.

On the other hand, how can I do what you mention in your message: 
And swich the python allocator to the c library by setting

PYTHONMALLOC=malloc

In the environment

Thanks for your response.

thiesmoeller, 2021-01-05:
if the images are not garbage collected, it seems, that some code of you still holds a reference to the list 'a'
Do you really have to hold 600 images in memory? or would it be possible to process them while grabbing?

For the grabbing, you don't have to use your own thread. pypylon already has an efficient c++ implementiation of a background grab loop -> see https://github.com/basler/pypylon/blob/master/samples/grabusinggrabloopthread.py

( the PYTHONMALLOC was only mentioned if you want to start to really debug your system, because the python3 pool memory allocator normally irritates memory debugging tools, and since 3.8 it is possible to disable it while debugging )

thiesmoeller, 2025-12-15:
This issue about memory accumulation requires more information about the specific code and memory profiling. Since we haven't received updates, we are closing this issue. If you are still experiencing memory issues, please open a new issue with detailed code and memory profiling information.