basler/pypylon #316: What is the code to change the frame rate?

egawatomoya, 2021-01-18:
Gain and exposure time can be set in "calltest.py" in ".\pypylon\tests\pylon_tests\usb", but frame rate settings cannot be found. What file is it written in? Or what kind of code can I set?

thiesmoeller, 2021-01-18:
pypylon gives you access to all features, that your camera has. 
Exactly as referenced in the manual.

https://docs.baslerweb.com/acquisition-frame-rate

the c++ code can really be converted by "search&replace"

so from a 
```c++
// Set the upper limit of the camera's frame rate to 30 fps
camera.AcquisitionFrameRateEnable.SetValue(true);
camera.AcquisitionFrameRate.SetValue(30.0);
```

you get:
```python
# Set the upper limit of the camera's frame rate to 30 fps
camera.AcquisitionFrameRateEnable.SetValue(True)
camera.AcquisitionFrameRate.SetValue(30.0)
```

Or in other words:
Configure your camera using pylon viewer. Learn about the parameters that you want to control.
Look up the documentation window in pylonviewer and you have your feature names :-)

[image]

kolupaev-hsw-de, 2022-03-22:
It doesn't work. Got an exception: "Node is not writable. : AccessException thrown in node 'AcquisitionFrameRateEnable' while calling 'AcquisitionFrameRateEnable.SetValue()' (file 'booleant.h', line 61)"

thiesmoeller, 2022-03-22:
Perhaps missing to Open your camera before?

`camera.Open()`

kolupaev-hsw-de, 2022-03-22:
@thiesmoeller, thank you, that helped.

closed 2022-03-23 by thiesmoeller