basler/pypylon #318: Basler USB3.0 ace 2 cameras wouldn't initialize again after running once until it is reset

sandeshk1, 2021-01-19:
Camera Model :  aca1920-160ucBAS

We use a docker service which ingests data from basler USB3.0 ace2 camera using a pylon driver. If the docker service is restarted or stopped and started again the camera fails to initialize. This behavior doesn't occur with ace series USB3.0 camera. The device has to be reset in order to be initialized again. Is there something different with ace2 series cameras or does it need any 

Note: camera is directly connected to the systems USB3.0 port.

sandeshk1, 2021-01-28:
Feature not writable in basler ace2 series USB3.0 camera after docker application ingesting from cameras is restarted.

sandeshk1, 2021-01-28:
Is there any script to reset basler USB3.0 cameras ?

lars-ek, 2021-02-14:
I have the problem, using Basler USB-cam on a jetson nano - if I reboot the nano without disconnecting the power, the camera is not accessible anymore. Therefore I was also searching for a reset function, but did not found anything... :(