basler/pypylon #319: Grabbing multiple cameras and opencv

marvision-ai, 2021-01-19:
Hello! 
I am having troubles modifying the capture multiple cameras example to do the following: 
1. setting each camera to have the following grab strategy: `cam.StartGrabbing(pylon.GrabStrategy_LatestImageOnly) ` 
2. capturing images from both cameras indefinitely
3. converting image data captured to opencv format and showing the images on 2 separate windows
4. having a way to only look at images from a particular camera on command instead of always going back and forth between the two i have.

Could you please help me add those features in? 

```python 
import os

os.environ["PYLON_CAMEMU"] = "3"

from pypylon import genicam
from pypylon import pylon
import sys
import cv2

converter = pylon.ImageFormatConverter()
# converting to opencv bgr format
converter.OutputPixelFormat = pylon.PixelType_BGR8packed
converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned

maxCamerasToUse = 2

# The exit code of the sample application.
exitCode = 0

try:

    # Get the transport layer factory.
    tlFactory = pylon.TlFactory.GetInstance()

    # Get all attached devices and exit application if no device is found.
    devices = tlFactory.EnumerateDevices()
    if len(devices) == 0:
        raise pylon.RuntimeException("No camera present.")

    # Create an array of instant cameras for the found devices and avoid exceeding a maximum number of devices.
    cameras = pylon.InstantCameraArray(min(len(devices), maxCamerasToUse))

    l = cameras.GetSize()

    # Create and attach all Pylon Devices.
    for i, cam in enumerate(cameras):
        cam.Attach(tlFactory.CreateDevice(devices[i])) 
        cam.StartGrabbing(pylon.GrabStrategy_LatestImageOnly) # PROBLEM
        # Print the model name of the camera.
        print("Using device ", cam.GetDeviceInfo().GetModelName())

    cameras.StartGrabbing()

    # Grab c_countOfImagesToGrab from the cameras.
    while cameras.IsGrabbing():
        grabResult = cameras.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)
        if grabResult.GrabSucceeded():

            cameraContextValue = grabResult.GetCameraContext()

            # Print the index and the model name of the camera.
            print("Camera ", cameraContextValue, ": ", cameras[cameraContextValue].GetDeviceInfo().GetModelName())

            # Now, the image data can be processed.

            # Access the image data
            # image = converter.Convert(grabResult)
            # img = image.GetArray()
            # cv2.namedWindow('cam1', cv2.WINDOW_NORMAL)
            # cv2.imshow('title', img)
            # k = cv2.waitKey(1)
            # if k == 27:
            #     break
            print("GrabSucceeded: ", grabResult.GrabSucceeded())
            print("SizeX: ", grabResult.GetWidth())
            print("SizeY: ", grabResult.GetHeight())
            img = grabResult.GetArray()
            print("Gray value of first pixel: ", img[0, 0])

except genicam.GenericException as e:
    # Error handling
    print("An exception occurred.", e.GetDescription())
    exitCode = 1

```

Thank you in advance :smile:

marvision-ai, 2021-01-19:
Update:

I have go the following code running and fully working. Is this optimal for realtime FPS? 

```python
import os

os.environ["PYLON_CAMEMU"] = "3"

from pypylon import genicam
from pypylon import pylon
import sys
import cv2

converter = pylon.ImageFormatConverter()
# converting to opencv bgr format
converter.OutputPixelFormat = pylon.PixelType_BGR8packed
converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned

maxCamerasToUse = 2

# The exit code of the sample application.
exitCode = 0

try:

    # Get the transport layer factory.
    tlFactory = pylon.TlFactory.GetInstance()

    # Get all attached devices and exit application if no device is found.
    devices = tlFactory.EnumerateDevices()
    if len(devices) == 0:
        raise pylon.RuntimeException("No camera present.")

    # Create an array of instant cameras for the found devices and avoid exceeding a maximum number of devices.
    cameras = pylon.InstantCameraArray(min(len(devices), maxCamerasToUse))

    l = cameras.GetSize()

    # Create and attach all Pylon Devices.
    for i, cam in enumerate(cameras):
        cam.Attach(tlFactory.CreateDevice(devices[i])) 
        #print(cam)
        #cam.StartGrabbing(pylon.GrabStrategy_LatestImageOnly) # PROBLEM
        # Print the model name of the camera.
        print("Using device ", cam.GetDeviceInfo().GetModelName())

    cameras.StartGrabbing(pylon.GrabStrategy_LatestImageOnly)

    # Grab c_countOfImagesToGrab from the cameras.
    while cameras.IsGrabbing():
        grabResult = cameras.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)
        if grabResult.GrabSucceeded():

            cameraContextValue = grabResult.GetCameraContext()

            # Print the index and the model name of the camera.
            #print("Camera ", cameraContextValue, ": ", cameras[cameraContextValue].GetDeviceInfo().GetModelName())

            # Now, the image data can be processed.
            #Access the image data
            image = converter.Convert(grabResult)
            img = image.GetArray()

            window = 'Camera-{}'.format(cameraContextValue)
            cv2.namedWindow(window, cv2.WINDOW_NORMAL)
            cv2.imshow(window, img)
            k = cv2.waitKey(1)
            if k == 27:
                break

except genicam.GenericException as e:
    # Error handling
    print("An exception occurred.", e.GetDescription())
    exitCode = 1
```

I assume that since the cameras are set to last_frame only, every time I display an image it is the most current frame correct? 

Also, I still cannot figure out how to attach more than 2 cameras and have it not run as an emulation device. I believe the USB3 bus is fully loaded with only 2 cameras. 

Please let me know your suggestions. Thanks!

thiesmoeller, 2021-01-20:
Hi @marvision-ai 

> only two camera on USB?

You can run a much higher number of cameras, depending on your system setup.
See this example where we demonstrate 16 USB cameras resulting in 1.3GB/s video transfer rate in to PC:
https://www.baslerweb.com/en/sales-support/knowledge-base/frequently-asked-questions/do-basler-cameras-use-the-full-usb-3-0-bandwidth/14862/

Your code will take the _first_ three cameras from the list of cameras discovered.
Depending on the sorting of the devices it is possible that an emulated camera ( simply disable them with PYLON_CAMEMU=0 ) is before your USB device.

> is the above code efficient

You have been recreating the opencv windows for every frame that you receive.
And an efficiency tip: If your camera has a BGR or RGB format nativly, please enable this. Image format conversion on your host is computationally expensive.

I reordered your code a bit:

```python
import os
import cv2
from pypylon import genicam
from pypylon import pylon

os.environ["PYLON_CAMEMU"] = "5"

converter = pylon.ImageFormatConverter()
# converting to opencv bgr format
converter.OutputPixelFormat = pylon.PixelType_BGR8packed
converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned

maxCamerasToUse = 5

# The exit code of the sample application.
exitCode = 0

try:

    # Get the transport layer factory.
    tlFactory = pylon.TlFactory.GetInstance()

    # Get all attached devices and exit application if no device is found.
    devices = tlFactory.EnumerateDevices()
    if len(devices) == 0:
        raise pylon.RuntimeException("No camera present.")

    # Create an array of instant cameras for the found devices and avoid exceeding a maximum number of devices.
    cameras = pylon.InstantCameraArray(min(len(devices), maxCamerasToUse))

    # Create and attach all Pylon Devices.
    for idx, cam in enumerate(cameras):
        cam.Attach(tlFactory.CreateDevice(devices[idx]))
        print("Using device ", cam.GetDeviceInfo().GetModelName())
        # setup named window per camera
        window_name = f'Camera-{idx:03}'
        cv2.namedWindow(window_name, cv2.WINDOW_NORMAL)

    cameras.StartGrabbing(pylon.GrabStrategy_LatestImageOnly)

    # Grab c_countOfImagesToGrab from the cameras.
    while cameras.IsGrabbing():
        grabResult = cameras.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)
        if grabResult.GrabSucceeded():

            cameraContextValue = grabResult.GetCameraContext()

            # Print the index and the model name of the camera.
            # print("Camera ", cameraContextValue, ": ", cameras[cameraContextValue].GetDeviceInfo().GetModelName())

            # Now, the image data can be processed.
            # Access the image data
            image = converter.Convert(grabResult)
            img = image.GetArray()

            window_name = f'Camera-{cameraContextValue:03}'
            cv2.imshow(window_name, img)
            k = cv2.waitKey(1)
            if k == 27:
                break

except genicam.GenericException as e:
    # Error handling
    print("An exception occurred.", e.GetDescription())
    exitCode = 1

```

thiesmoeller, 2021-01-20:
We use the pylon camera emulator in most of the examples, as it allows to verify application code without a camera connected.
So in real application you could just leave out setting PYLON_CAMEMU in the environment.
Details in the docs "Camera Emulation | Basler" https://docs.baslerweb.com/camera-emulation

configure PixelFormat accordingly per camera. Read the camera documentation.

To verify overall camera functionality please always use pylon viewer on your target first. 
Pylon viewer will show you code snippets for each feature so it is easy to learn what to do in pypylon.

Do you see all camera when calling lsusb ?

Gururaz, 2022-10-18:
HI @thiesmoeller 
Thank you for contributing. The code works fine.

1. I am able to stream both cameras in real time but I see the FPS drops for one of the cameras in my case. I am not clear on this part.

2. I am new to working with this library. I am interested in saving the images separately into a folder with the images taken from the left and the right camera upon clicking key "s" during the while loop "while cameras.IsGrabbing():" is getting executed.  But I cannot interpret the pixel value obtained from the "img = image.GetArray()". Is it from the right camera or the left camera? 

It would be great if you can guide me with your knowledge. 

Thank you for your time,

thiesmoeller, 2025-12-15:
This issue about grabbing with multiple cameras and OpenCV has been addressed in the comments with code examples. We are closing this issue. For further questions on multiple cameras, please refer to the discussion or open a new issue.